amd/xdna-driver #2: Files missing from debian package?

spaceotter, 2024-01-26:
I was able to build everything, but the post-install step breaks:
```
Successfully intalled and enabled DKMS for xrt-amdxdna/2.17.0
Loading new amdxdna Linux kernel module
Creating xclbin firmware symbolic link
/opt/xilinx/xrt/amdxdna/setup_xclbin_firmware.sh: line 38: cd: /lib/firmware/amdipu/*/: No such file or directory
modprobe: ERROR: could not insert 'amdxdna': Unknown symbol in module, or unknown parameter (see dmesg)
```

```
$ sudo dkms status
xrt/2.17.0: added
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/xrt-amdaie/2.17.0/source/dkms.conf does not exist.
```

maxzhen, 2024-01-26:
1. Can you manually create /lib/firmware/amdipu and try again?
2. Did you build and installed the 6.7 kernel as in the README?

spaceotter, 2024-01-26:
Ah sorry, yes I built and installed the 6.7 kernel based on the README. That part seemed to go well but the instructions are a little off.

spaceotter, 2024-01-26:
Creating the folder /lib/firmware/amdipu doesn't change the output.

spaceotter, 2024-01-26:
I figured out the missing firmware issue. The build script expects system package `jq` to be installed, and doesn't report an error if the firmware download fails.

maxzhen, 2024-01-26:
Will update readme to make sure user will install jq before build.

maxzhen, 2024-01-29:
Fixed by PR#4.

closed 2024-01-29 by maxzhen