amd/xdna-driver #5: How about integration with Pytorch?

serhii-nakon, 2024-01-27:
I think that integration with Pytorch can trigger to buy those laptops just to start working with AI. 
Also it can be opponent to Macbooks that already use their iGPU for Pytorch tasks and it very popular usage case!!!

And need to remember that Ryzen has dedicated AI engine that more power efficient then Macbook's iGPU that a big pros!

ntua-el19019, 2024-01-28:
I agree with that, It is not even a matter of choice anymore if you want to integrate local AI tasks such as LLMs. For example existing frameworks that allow you to run models even in you CPU efficiently, in the case of ollama, would greatly benefit from the acceleration of an NPU.

maxzhen, 2024-01-29:
There will be other common software layers built upon the driver / XRT to do this.

serhii-nakon, 2024-01-29:
Do you have some plans?

serhii-nakon, 2024-01-29:
@maxzhen Do you have some plans?

andyluo7, 2024-01-31:
@serhii-nakon , you can export from pytorch to onnx models and run it with Ryzen AI SW to leverage NPU in Ryzen AI laptops. For more info, please check - https://ryzenai.docs.amd.com/en/latest/

closed 2024-01-31 by maxzhen

serhii-nakon, 2024-01-31:
@andyluo7 I mean use NPU to accelerate ML, not only execution!
Like use it instead of GPU fully.

andyluo7, 2024-01-31:
@serhii-nakon, you can use NPU to accelerate ML inference following https://ryzenai.docs.amd.com/en/latest/. NPU is not designed for ML training.

nstiurca, 2024-06-27:
> @serhii-nakon , you can export from pytorch to onnx models and run it with Ryzen AI SW to leverage NPU in Ryzen AI laptops. For more info, please check - [ryzenai.docs.amd.com/en/latest](https://ryzenai.docs.amd.com/en/latest/)

Ryzen AI is for Windows 11, per its [installation instructions](https://ryzenai.docs.amd.com/en/latest/inst.html).

So I think the question remains how to use this _Linux_ driver for NPU acceleration of Pytorch in _Linux_?