amd/xdna-driver #8: Is possible to develop a custom AI Engine firmware?

kele14x, 2024-01-30:
I notice the Ryzen AI are using XLNX_VART_FIRMWARE and .xclbin file. I expect it was using the same technology as Xilinx Versal AI Engine. Will AMD open the document and interface for users to develop custom firmware? (That is, develop custom .xclbin file just like Versal device using Vitis IDE 
😄). This is great for non-AI applications to utilize the AI Engine inside Ryzen CPU.

andyluo7, 2024-01-31:
@kele14x , there is an open source project from AMD Research team about creating custom AIE kernel. For more details, please check https://github.com/AMDResearch/Riallto/blob/main/notebooks/4_2_write_your_kernel.ipynb

closed 2024-01-31 by maxzhen

jummama, 2024-02-19:
I believe @kele14x was asking about the possibility of using the AI engine as a more general purpose FPGA core via a custom firmware, rather than only being able to use it for it's intended purpose as an AI coprocessor.

As I understand it, the Ryzen AI engine is essentially "just" an FPGA embedded into newer Ryzen chips, along with a firmware to set it up for AI processing tasks.

Unless I'm missing something, the page you linked discusses how to leverage the functionality as officially intended, using the AMD-supplied bitstream (firmware). This is certainly good info, but wouldn't help with lower level FPGA experimentation.

One example I could think of is setting up an ARM or RISC-V core for more efficient virtualization/emulation, or utilizing it for customized encryption acceleration.

maxzhen, 2024-02-19:
There is no FPGA there. It's hard-IP.

kele14x, 2024-02-21:
I will be very happy if AMD embedded FPGA into new Ryzen chips. But sadly from the document they only copied AI Engine from Versal into Ryzen. Hope they can develop a real "all in one" chip in further :)

Joel-De, 2024-02-22:
> @kele14x , there is an open source project from AMD Research team about creating custom AIE kernel. For more details, please check https://github.com/AMDResearch/Riallto/blob/main/notebooks/4_2_write_your_kernel.ipynb

Is this known to work on Linux using this driver? When attempting to run notebooks from that repository I run into issues where a hard reference is made to a powershell script that obviously fails to run on Linux when importing the provided npu library.

Could entirely be (and probably is) an issue on my end - but I'd like to confirm that it is supposed to work in the first place on Linux.

kele14x, 2024-02-22:
I think the `xdna-driver` (this repo) surely works on Linux since it's Linux driver. But for the Riallto project and the notebooks, are designed work at Windows. I hope there is way to run it under Linux too.

AEW2015, 2024-04-17:
I will just add that the Vitis GUI provides a powerful tool to design and debug AIE-ML workloads.  It seems silly if you are already using the tool to provide compiling for the system to not give designers access to the GUI for it.

Sfinx, 2024-04-19:
> There is no FPGA there. It's hard-IP.

Then why the xclbin file is needed there ? What it contains ?

keryell, 2024-04-19:
> Then why the xclbin file is needed there ? What it contains ?

The `xclbin` evolved into a more generic container than the . It can contains the ELF for the AIE cores and so on.
To get an idea, you can look at some mechanics in https://github.com/Xilinx/mlir-aie/blob/main/python/compiler/aiecc/main.py and more generally the whole project.

Sfinx, 2024-04-20:
Found xclbinutil, it shows for xclbinutil --info --input tools/bins/1502_00/validate.xclbin:

```
Generated By
------------
   Command:       v++
   Version:       2021.2.0 - 2021-05-17-22:18:13 (SW BUILD: 0)
   Command Line:  v++ --input_files _x.hw_emu.xilinx_v1_ipu_202020_1/vadd.xo --link --optimize 0 --output ./build_dir.hw_emu.xilinx_v1_ipu_202020_1/vadd.link.xclbin --platform xilinx_v1_ipu_202020_1 --report_level 0 --save-temps --target hw_emu --temp_dir ./_x.hw_emu.xilinx_v1_ipu_202020_1 --user_ip_repo_paths /wrk/xhdhdnobkup2/madhusud/CRs/hello_world_xrt/ip_repo 
   Options:       --input_files _x.hw_emu.xilinx_v1_ipu_202020_1/vadd.xo
                  --link
                  --optimize 0
                  --output ./build_dir.hw_emu.xilinx_v1_ipu_202020_1/vadd.link.xclbin
                  --platform xilinx_v1_ipu_202020_1
                  --report_level 0
                  --save-temps
                  --target hw_emu
                  --temp_dir ./_x.hw_emu.xilinx_v1_ipu_202020_1
                  --user_ip_repo_paths /wrk/xhdhdnobkup2/madhusud/CRs/hello_world_xrt/ip_repo 
```

Okay, seems like path is :

- create C++ function
- compile it with Vitis, get the .xo file
- create xclbin file with v++ for xilinx_v1_ipu_202020_1 kernel
- load & run xclbin

Where I can find the source of the xilinx_v1_ipu_202020_1/vadd.xo that is embedded in validate.xclbin ?