amd/xdna-driver #18: Problems with suspend and hibernate

keryell, 2024-02-16:
While I have a lot of issues with power control on my laptop with 6.5 and 6.7.4 kernel independently from this project, it seems that sometimes the `amdxdna` prevents the kernel from stopping too.
When it works:
```
2024-02-16T09:50:10.661313-08:00 rk-xsj kernel: [  139.176934] amdxdna 0000:66:00.1: firmware resuming...
2024-02-16T09:50:10.661314-08:00 rk-xsj kernel: [  139.177070] amdxdna 0000:66:00.1: hardware context resuming...
```
When it fails:
```
2024-02-16T10:00:26.051836-08:00 rk-xsj kernel: [  668.382497] amdxdna 0000:66:00.1: amdxdna_do_suspend: suspend NPU firmware failed
2024-02-16T10:00:26.051837-08:00 rk-xsj kernel: [  668.382501] amdxdna 0000:66:00.1: PM: pci_pm_suspend(): amdxdna_pmops_suspend+0x0/0x80 [amdxdna] returns -19
2024-02-16T10:00:26.051837-08:00 rk-xsj kernel: [  668.382514] amdxdna 0000:66:00.1: PM: dpm_run_callback(): pci_pm_suspend+0x0/0x1b0 returns -19
2024-02-16T10:00:26.051838-08:00 rk-xsj kernel: [  668.382519] amdxdna 0000:66:00.1: PM: failed to suspend async: error -19
2024-02-16T10:00:26.051838-08:00 rk-xsj kernel: [  668.478472] PM: Some devices failed to suspend, or early wake event detected
```
Is this a known problem?

maxzhen, 2024-02-16:
We also see the issues with suspend / resume. Not exactly the same, but it does not work well. I have updated the code today with some changes and will be updating it again very soon some additional fix. You could give it a try later and see if it fixes your problem.

maxzhen, 2024-02-19:
PR#20 contains the further fix. Suspend / resume should work by now.

closed 2024-02-19 by maxzhen

keryell, 2024-02-20:
Thanks. It works with `s2ram` and `s2disk`.

maxzhen, 2024-02-20:
Thanks for testing!