amd/xdna-driver #33: UBSAN: array-index-out-of-bounds in /var/lib/dkms/xrt-amdxdna/2.17.0/build/driver/amdxdna/npu1_message.c:488:57

keryell, 2024-04-03:
With current driver I can run
```
-<%>- ./example_build/example_noop_test /lib/firmware/amdnpu/1502_00/validate.xclbin                 
Host test code start...
Host test code is creating device object...
Host test code is loading xclbin object...
Host test code is creating kernel object...
Host test code kernel name: DPU_PDI_0
Host code is registering xclbin to the device...
Host code is creating hw_context...
Host test code is creating kernel object...
Host test code allocate buffer objects...
Host test code sync buffer objects to device...
Host test code iterations (~10 seconds): 70000
Host test microseconds: 6833094
Host test average latency: 97 us/iter
TEST PASSED!
```
But I have this kernel error message:
```
[Wed Apr  3 13:18:21 2024] [drm] Initialized amdxdna_accel_driver 1.0.0 20240124 for 0000:66:00.1 on minor 0
[Wed Apr  3 13:19:20 2024] ------------[ cut here ]------------
[Wed Apr  3 13:19:20 2024] UBSAN: array-index-out-of-bounds in /var/lib/dkms/xrt-amdxdna/2.17.0/build/driver/amdxdna/npu1_message.c:488:57
[Wed Apr  3 13:19:20 2024] index 1 is out of range for type 'amdxdna_cu_config [*]'
[Wed Apr  3 13:19:20 2024] CPU: 3 PID: 27749 Comm: example_noop_te Tainted: P        W  OE      6.8.2+iommu-sva-part4-v7+pmc-10ms-delay+ #2
[Wed Apr  3 13:19:20 2024] Hardware name: HP HP ZBook Power 15.6 inch G10 A Mobile Workstation PC/8B95, BIOS V85 Ver. 01.04.00 01/18/2024
[Wed Apr  3 13:19:20 2024] Call Trace:
[Wed Apr  3 13:19:20 2024]  <TASK>
[Wed Apr  3 13:19:20 2024]  dump_stack_lvl+0x48/0x70
[Wed Apr  3 13:19:20 2024]  dump_stack+0x10/0x20
[Wed Apr  3 13:19:20 2024]  __ubsan_handle_out_of_bounds+0xc6/0x110
[Wed Apr  3 13:19:20 2024]  npu1_config_cu+0x36a/0x3e0 [amdxdna]
[Wed Apr  3 13:19:20 2024]  ? __pfx_npu_msg_cb+0x10/0x10 [amdxdna]
[Wed Apr  3 13:19:20 2024]  npu1_hwctx_config+0xa4/0x200 [amdxdna]
[Wed Apr  3 13:19:20 2024]  amdxdna_drm_config_hwctx_ioctl+0xa4/0x140 [amdxdna]
[Wed Apr  3 13:19:20 2024]  ? __pfx_amdxdna_drm_config_hwctx_ioctl+0x10/0x10 [amdxdna]
[Wed Apr  3 13:19:20 2024]  drm_ioctl_kernel+0xb9/0x120
[Wed Apr  3 13:19:20 2024]  ? srso_alias_return_thunk+0x5/0xfbef5
[Wed Apr  3 13:19:20 2024]  drm_ioctl+0x2d4/0x550
[Wed Apr  3 13:19:20 2024]  ? __pfx_amdxdna_drm_config_hwctx_ioctl+0x10/0x10 [amdxdna]
[Wed Apr  3 13:19:20 2024]  __x64_sys_ioctl+0xa0/0xf0
[Wed Apr  3 13:19:20 2024]  do_syscall_64+0x74/0x140
[Wed Apr  3 13:19:20 2024]  ? srso_alias_return_thunk+0x5/0xfbef5
[Wed Apr  3 13:19:20 2024]  ? sysvec_apic_timer_interrupt+0x4b/0xd0
[Wed Apr  3 13:19:20 2024]  entry_SYSCALL_64_after_hwframe+0x6e/0x76
[Wed Apr  3 13:19:20 2024] RIP: 0033:0x751c6a12396f
[Wed Apr  3 13:19:20 2024] Code: 00 48 89 44 24 18 31 c0 48 8d 44 24 60 c7 04 24 10 00 00 00 48 89 44 24 08 48 8d 44 24 20 48 89 44 24 10 b8 10 00 00 00 0f 05 <89> c2 3d 00 f0 ff ff 77 18 48 8b 44 24 18 64 48 2b 04 25 28 00 00
[Wed Apr  3 13:19:20 2024] RSP: 002b:00007ffc7c102470 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[Wed Apr  3 13:19:20 2024] RAX: ffffffffffffffda RBX: 00000000c0186442 RCX: 0000751c6a12396f
[Wed Apr  3 13:19:20 2024] RDX: 00007ffc7c1026b0 RSI: 00000000c0186442 RDI: 0000000000000003
[Wed Apr  3 13:19:20 2024] RBP: 00007ffc7c1026b0 R08: 0000751c641e0000 R09: 0000000000000003
[Wed Apr  3 13:19:20 2024] R10: 0000598033302d40 R11: 0000000000000246 R12: 00005980332d69e0
[Wed Apr  3 13:19:20 2024] R13: 000059803331ba28 R14: 0000598033302c50 R15: 0000751c6aa72660
[Wed Apr  3 13:19:20 2024]  </TASK>
[Wed Apr  3 13:19:20 2024] ---[ end trace ]---
```

keryell, 2024-04-03:
But perhaps instead of debugging this it would be more useful to port the XDNA driver to the latest XRT and see how it goes from there?

maxzhen, 2024-04-03:
It is a driver bug. Not in XRT.

mamin506, 2024-04-03:
I cannot reproduce the issue on my Phoenix board with the latest commit  37961e13d1e78a59f7231e8c37b6c8735f2c0281. 

@keryell , can you please share below information?
$ xbutil examine -d
$ uname -a
$ md5sum /lib/firmware/amdnpu/1502_00/validate.xclbin

mamin506, 2024-04-03:
This is a warning. Actually, driver doesn't access out of range. So, it is harmless.

The root cause is that we defined a flexible array member [here](https://github.com/amd/xdna-driver/blob/main/src/include/uapi/drm_local/amdxdna_accel.h#L131). It looks like,
 
struct amdxdna_hwctx_param_config_cu {
	__u16 num_cus;
	__u16 pad[3];
	struct amdxdna_cu_config cu_configs[0];
};

If the Linux is config with UBSAN and it is enabled in Linux cmdline. User will see this warning.

I will provide a fix soon. For now, this is safe to ignore.

keryell, 2024-04-04:
Great you have found the issue.
This is not a flexible array but an array with 0 element.
I think this should be instead:
```c
	struct amdxdna_cu_config cu_configs[];
```

mamin506, 2024-04-04:
> Great you have found the issue. This is not a flexible array but an array with 0 element. I think this should be instead:
> 
> ```c
> 	struct amdxdna_cu_config cu_configs[];
> ```

The fix will be,
``` c
    struct amdxdna_cu_config cu_configs[] __counted_by(num_cus);
```
The __counted_by() is used on flexible array member. It will provide information to bounds sanitizer as well. So that if driver do access out of range, UBSAN can still catch it. At that time, it will be a real driver bug.

keryell, 2024-04-08:
> The __counted_by() is used on flexible array member. It will provide information to bounds sanitizer as well. So that if driver do access out of range, UBSAN can still catch it. At that time, it will be a real driver bug.

Even better! This is a motivation to quickly adopt gcc-14 and Clang-18 to benefit from this.
Thanks.

keryell, 2024-04-12:
@mamin506 thanks, it looks you fixed this in https://github.com/amd/xdna-driver/pull/34 along a few others too. :smile:

closed 2024-04-12 by keryell