amd/xdna-driver #43: clinfo outputs the '[XRT] ERROR: get_device_info: Operation not supported'

Sfinx, 2024-04-17:
Should it work ?

```
clinfo 
Number of platforms:				 2
  Platform Profile:				 EMBEDDED_PROFILE
  Platform Version:				 OpenCL 1.0
  Platform Name:				 Xilinx
  Platform Vendor:				 Xilinx
  Platform Extensions:				 cl_khr_icd
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 2.1 AMD-APP (3558.0)
  Platform Name:				 AMD Accelerated Parallel Processing
  Platform Vendor:				 Advanced Micro Devices, Inc.
  Platform Extensions:				 cl_khr_icd cl_amd_event_callback 

  Platform Name:				 Xilinx
Number of devices:				 1
XRT build version: 2.17.0
Build hash: baf88820fb3fc24dda4dc08c91ecbca2c76c7b0f
Build date: 2024-04-17 13:03:42
Git branch: HEAD
PID: 61787
UID: 0
[Wed Apr 17 16:03:16 2024 GMT]
HOST: Shiva
EXE: /opt/rocm-5.5.0/bin/clinfo
[XRT] ERROR: get_device_info: Operation not supported
ERROR: clGetDeviceInfo(-6)
```
It is AMD Ryzen 7 7840HS:

```
xbutil examine
System Configuration
  OS Name              : Linux
  Release              : 6.8.6-060806-generic
  Version              : #202404131135 SMP PREEMPT_DYNAMIC Wed Apr 17 04:32:21 EEST 2024
  Machine              : x86_64
  CPU Cores            : 16
  Memory               : 95772 MB
  Distribution         : Ubuntu 22.04.4 LTS
  GLIBC                : 2.35
  Model                : TUXEDO Sirius 16 Gen1
  BIOS vendor          : American Megatrends International, LLC.
  BIOS version         : V1.00A00_20240108

XRT
  Version              : 2.17.0
  Branch               : HEAD
  Hash                 : baf88820fb3fc24dda4dc08c91ecbca2c76c7b0f
  Hash Date            : 2024-04-17 13:03:42
  XOCL                 : 2.17.0, baf88820fb3fc24dda4dc08c91ecbca2c76c7b0f
  XCLMGMT              : 2.17.0, baf88820fb3fc24dda4dc08c91ecbca2c76c7b0f
  AMDXDNA              : 2.17.0_20240417, 35351e4bbbc65568669c36255825425030be721f

Devices present
BDF             :  Name          
---------------------------------
[0000:6a:00.1]  :  RyzenAI-npu1  
```

maxzhen, 2024-04-17:
We currently do not support non-XRT API flows. But we can look into it later.

Sfinx, 2024-04-17:
Then this have to be removed:

```
cat /etc/OpenCL/vendors/xilinx.icd 
/opt/xilinx/xrt/lib/libxilinxopencl.so
```

maxzhen, 2024-04-17:
That is for Alveo devices that XRT supports. We just do not support OpenCL for NPU devices yet.

Sfinx, 2024-04-17:
I see

closed 2024-04-25 by maxzhen