amd/xdna-driver #68: Build XRT failed. Any depends on Vitis?

alexhegit, 2024-05-06:
[image]

Got issue 

[image]

What dose it means? any depends on Vitis?

maxzhen, 2024-05-06:
You can ignore this warning message. I do not see other cmake output from your log. I'm assuming they work for you.

keryell, 2024-05-07:
Perhaps there are some missing requirements which are not available on your system?
Like `sudo xrt/src/runtime_src/tools/scripts/xrtdeps.sh` mentioned in https://xilinx.github.io/XRT/master/html/build.html ?

mamin506, 2024-05-16:
> Perhaps there are some missing requirements which are not available on your system? Like `sudo xrt/src/runtime_src/tools/scripts/xrtdeps.sh` mentioned in https://xilinx.github.io/XRT/master/html/build.html ?

In this repo's readme, see https://github.com/amd/xdna-driver?tab=readme-ov-file#prerequisite
User should run amdxdna_deps.sh with root privilege. Under the hood, it will run xrtdeps.sh you mentioned.

The xdna-driver doesn't depend on Vitis tool.
In this case, XRT submodule is warning people that it will NOT build some MicroBlaze firmware. The Ryzen AI doesn't require those binaries. As Max said, please ignore these warning.

This warning doesn't lead to XRT build failed. If XRT build failed, we will need full build log to tell why.

antheas, 2024-06-02:
Seems like you need to run build in XRT like this:
`./build.sh -noert`

As the error says. Not that the build worked... But at the same time I am not on Ubuntu.

maxzhen, 2024-06-02:
> Seems like you need to run build in XRT like this: `./build.sh -noert`
> 
> As the error says. Not that the build worked... But at the same time I am not on Ubuntu.

You can do build with '-noert', but that does not matter, as we have suggested above. Please provide full build log for us to see why it fail. Ubuntu might be a key requirement here, yes.

antheas, 2024-06-03:
I rebuilt in Ubuntu to try it and it worked.

Without using the -noert option, it just displays the error and exits. I also had to do it in Ubuntu. Maybe it changed in a recent version.

I would consider updating the build instructions.

maxzhen, 2024-06-21:
Done: https://github.com/amd/xdna-driver/pull/132

closed 2024-06-21 by maxzhen