amd/xdna-driver #103: Kernel 6.9 released

tw-atroehrsm, 2024-05-23:
Hello,

question. Does Kernel 6.9 include the necessary patches (IOMMU SVA) or does it still require a custom kernel?

Thx

nstiurca, 2024-06-01:
I've been trying to track when this is mainlined as well, and from what I can tell 6.10 is the most likely version to contain all the necessary patches. In fact I think last week's 6.10-rc1 (probably) has all these patches.

Part 4 (of 4) of the patch series (ie, the patches listed above) was [applied on Apr 18](https://lore.kernel.org/linux-iommu/ZiuIneLj690wcrIU@8bytes.org/) and got pulled into Linus's tree [about 2 weeks ago](https://lore.kernel.org/linux-iommu/171605562141.7011.17834859902168748093.pr-tracker-bot@kernel.org/#t).

maxzhen, 2024-06-01:
Yes, 6.10 should have all the patches we need. We're internally testing 6.10. We'll update this repo when we complete the test.

closed 2024-06-01 by maxzhen

garloff, 2024-06-01:
Cherry-picking
```
b194fc01fbaf5 platform/x86/amd/pmc: Add 10ms delay to 2nd s0i3 cycle for all APUs
1cf1c512d240d Add compilation recipe for AMD IOMMU SVA branches
39ecf66190804 Better describe branches for AMD IOMMU SVA
591b43ada1659 Add AMD-specific README about RyzenAI an IOMMU SVA
759eb0ce6369d iommu/amd: Add SVA domain support
811fefdb25a1f iommu: Add ops->domain_alloc_sva()
0aa515afdb098 iommu/amd: Initial SVA support for AMD IOMMU
95d5d54b96628 iommu/amd: Add support for enable/disable IOPF
5ec1dfbdcf9f6 iommu/amd: Add IO page fault notifier handler
5a28ef6a108f8 iommu/amd: Add support for page response
fa8b5564e1a6b iommu/amd: Define per-IOMMU iopf_queue
d6650f92fee8f iommu/amd: Enable PCI features based on attached domain capability
4008d455c1a61 iommu/amd: Setup GCR3 table in advance if domain is SVA capable
82bc8f69d14c8 iommu/amd: Introduce iommu_dev_data.max_pasids
4d12b7d853f02 iommu/amd: Fix PPR interrupt processing logic
af2a86616ca3b iommu/amd: Move PPR-related functions into ppr.c
f648e97453492 iommu/amd: Add support for enabling/disabling IOMMU features
da271b4b64ad0 iommu/amd: Introduce per device DTE update function
b213215e9ea7d iommu/amd: Rename amd_iommu_v2_supported() as amd_iommu_pasid_supported()
a1130eba6fb6b PART4: SVA support (SVA + IOPF)
```
and commenting out the line 196 in `drivers/iommu/amd/ppr.c`:
```C
                //if (!amd_iommu_snp_en)
                        raw[0] = raw[1] = 0UL;
```
to make it compile was successful for me.
```shell
garloff@framekurt(//):~ [0]$ uname -a
Linux framekurt 6.9.3-sva4-KG+ #2 SMP PREEMPT_DYNAMIC Sat Jun  1 09:02:58 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
garloff@framekurt(//):~ [0]$ dmesg | grep -C1 amdxdna
[    3.798179] ccp 0000:c1:00.2: psp enabled
[    3.800229] amdxdna: loading out-of-tree module taints kernel.
[    3.802884] amd-tee driver initialization successful
[    3.805391] hid-sensor-hub 0018:32AC:001B.0002: hidraw1: I2C HID v1.00 Device [FRMW0005:00 32AC:001B] on i2c-FRMW0005:00
[    3.847207] amdxdna 0000:c2:00.1: (Develop) IOMMU mode is 0
[    3.882695] mc: Linux media interface: v0.10
[    3.885573] amdxdna 0000:c2:00.1: set mpnpu_clock = 600 mhz
[    3.886475] cros_ec_lpcs cros_ec_lpcs.0: loaded with quirks 00000001
[    3.896724] cros_ec_lpcs cros_ec_lpcs.0: Chrome EC device registered
[    3.904573] amdxdna 0000:c2:00.1: set npu_hclock = 1024 mhz
[    3.907055] RAPL PMU: API unit is 2^-32 Joules, 1 fixed counters, 163840 ms ovfl timer
--
[    3.918260] amd-pmf AMDI0102:00: registered PMF device successfully
[    3.945704] [drm] Initialized amdxdna_accel_driver 1.0.0 20240124 for 0000:c2:00.1 on minor 0
[    3.953256] Adding 8388604k swap on /dev/nvme0n1p3.  Priority:41 extents:1 across:8388604k SS
```
If we can work with an unpatched upstream 6.10 kernel, that would of course be better.