amd/xdna-driver #118: DRM_IOCTL_AMDXDNA_GET_INFO IOCTL failed (err=95): Operation not supported

firstwaterer, 2024-06-13:
```
# xrt_test

====== 0: npu3 xrt vadd started =====
DRM_IOCTL_AMDXDNA_GET_INFO IOCTL failed (err=95): Operation not supported
====== 0: npu3 xrt vadd FAILED  =====

1       test(s) executed
1       test(s) FAILED!
```
```
# xbutil validate -d 0000:c7:00.1

XRT build version: 2.17.0
Build hash:
Build date: 2024-06-10 20:36:38
Git branch:
PID: 2214
UID: 0
[Thu Jun 13 10:05:29 2024 GMT]
HOST: m-kot
EXE: /usr/bin/unwrapped/xbutil2
[xbutil] ERROR: DRM_IOCTL_AMDXDNA_GET_INFO IOCTL failed (err=95): Operation not supported
```
- XRT version:
```202410.2.17.319``` [https://github.com/Xilinx/XRT/releases/tag/202410.2.17.319](url)
- platform:
```AMD Ryzen™ 7 8845HS```
- dmesg
```
# dmesg | grep -v input | grep -E "xdna|npu|xocl|xclmgmt|drm|gpu"
[    1.347323] xocl: loading out-of-tree module taints kernel.
[    1.356150] xclmgmt init()
[    2.554377] systemd[1]: Starting Load Kernel Module drm...
[    2.560314] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    2.560358] systemd[1]: Finished Load Kernel Module drm.
[    2.801077] Loading firmware: amdnpu/1502_00/npu.sbin
[    2.802016] amdxdna 0000:c7:00.1: enabling device (0000 -> 0002)
[    2.809259] amdxdna 0000:c7:00.1: (Develop) IOMMU mode is 0
[    2.840447] amdxdna 0000:c7:00.1: set mpnpu_clock = 600 mhz
[    2.860449] amdxdna 0000:c7:00.1: set npu_hclock = 1024 mhz
[    2.894357] [drm] amdgpu kernel modesetting enabled.
[    2.897619] amdgpu: Virtual CRAT table created for CPU
[    2.897626] amdgpu: Topology: Add CPU node
[    2.897713] amdgpu 0000:c6:00.0: enabling device (0006 -> 0007)
[    2.897734] [drm] initializing kernel modesetting (IP DISCOVERY 0x1002:0x1900 0x1002:0x0124 0xC5).
[    2.897741] [drm] register mmio base: 0xDC500000
[    2.897741] [drm] register mmio size: 524288
[    2.900727] [drm] add ip block number 0 <soc21_common>
[    2.900730] [drm] add ip block number 1 <gmc_v11_0>
[    2.900732] [drm] add ip block number 2 <ih_v6_0>
[    2.900734] [drm] add ip block number 3 <psp>
[    2.900736] [drm] add ip block number 4 <smu>
[    2.900737] [drm] add ip block number 5 <dm>
[    2.900739] [drm] add ip block number 6 <gfx_v11_0>
[    2.900740] [drm] add ip block number 7 <sdma_v6_0>
[    2.900742] [drm] add ip block number 8 <vcn_v4_0>
[    2.900744] [drm] add ip block number 9 <jpeg_v4_0>
[    2.900745] [drm] add ip block number 10 <mes_v11_0>
[    2.900761] amdgpu 0000:c6:00.0: amdgpu: Fetched VBIOS from VFCT
[    2.900764] amdgpu: ATOM BIOS: 113-PHXGENERIC-001
[    2.900777] Loading firmware: amdgpu/psp_13_0_4_toc.bin
[    2.901513] [drm] Initialized amdxdna_accel_driver 1.0.0 20240124 for 0000:c7:00.1 on minor 0
[    2.901597] Loading firmware: amdgpu/psp_13_0_4_ta.bin
[    2.902331] Loading firmware: amdgpu/dcn_3_1_4_dmcub.bin
[    2.903122] Loading firmware: amdgpu/gc_11_0_1_pfp.bin
[    2.903583] Loading firmware: amdgpu/gc_11_0_1_me.bin
[    2.904080] Loading firmware: amdgpu/gc_11_0_1_rlc.bin
[    2.904605] Loading firmware: amdgpu/gc_11_0_1_mec.bin
[    2.905238] Loading firmware: amdgpu/gc_11_0_1_imu.bin
[... 94 lines cut ...]
[  314.815929] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816025] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816115] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816209] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816299] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816386] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816482] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816573] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816671] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816766] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816858] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.816951] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  314.817045] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  370.301712] amdxdna 0000:c7:00.1: set mpnpu_clock = 600 mhz
[  370.321703] amdxdna 0000:c7:00.1: set npu_hclock = 1024 mhz
[  370.363371] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  891.645155] amdxdna 0000:c7:00.1: set mpnpu_clock = 600 mhz
[  891.661209] amdxdna 0000:c7:00.1: set npu_hclock = 1024 mhz
[  891.702813] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  900.371245] amdxdna 0000:c7:00.1: set mpnpu_clock = 600 mhz
[  900.391240] amdxdna 0000:c7:00.1: set npu_hclock = 1024 mhz
[  900.433023] amdxdna 0000:c7:00.1: aie2_get_info: Not supported request parameter 7
[  906.238892] amdxdna 0000:c7:00.1: set mpnpu_clock = 600 mhz
[  906.258981] amdxdna 0000:c7:00.1: set npu_hclock = 1024 mhz
```

maxzhen, 2024-06-13:
NPU3 platform is not yet supported.

closed 2024-06-13 by maxzhen

firstwaterer, 2024-06-13:
But the previous build works, like "xbutil validate -d 0000:c7:00.1" which validates correctly.

firstwaterer, 2024-06-13:
> NPU3 platform is not yet supported.

The code requesting DRM_IOCTL_AMDXDNA_GET_INFO doesn't seem to exist before commit 7682e0b4b6c435d6700faef3dc27cd709324e27f right?

maxzhen, 2024-06-14:
Were you testing the latest driver in this repo? I thought we have moved to XRT version 2.18. But you're running XRT 2.17? How did you build and install XRT on your system?

firstwaterer, 2024-06-14:
> Were you testing the latest driver in this repo? I thought we have moved to XRT version 2.18. But you're running XRT 2.17? How did you build and install XRT on your system?

The xdna-driver is the latest version. Packaged with portage, and changed its installation path with some patches.
In the build, xdna-driver's libxrt_core library and libxrt_coreutils would conflict with the installed XRT libraries(version 2.17), so they were removed and xrt 2.17 [https://github.com/Xilinx/XRT/tree/202410.2.17.319](url) was used.

maxzhen, 2024-06-14:
Please follow instructions listed in the readme and build XRT from the submodule directory. It does not work with some random version of XRT.