amd/xdna-driver #121: `example_noop_test` buffer overflows

makslevental, 2024-06-14:
```
(base) mlevental@mlevental-F7BSC:/tmp/xdna-driver/build$ ./example_build/example_noop_test ../tools/bins/1502_00/validate.xclbin
Host test code start...
Host test code is creating device object...
Host test code is loading xclbin object...
Host test code is creating kernel object...
Host test code kernel name: DPU_PDI_0
Host code is registering xclbin to the device...
Host code is creating hw_context...
Host test code is creating kernel object...
Host test code allocate buffer objects...
Host test code sync buffer objects to device...
Host test code iterations (~10 seconds): 70000
*** stack smashing detected ***: terminated
Aborted (core dumped)
```

```
(base) mlevental@mlevental-F7BSC:/tmp/xdna-driver/build$ git rev-parse HEAD^
85e380b68c8b921b7efab18c1b3280644a982d40
```

```
(base) mlevental@mlevental-F7BSC:/tmp/xdna-driver/build$ /opt/xilinx/xrt/bin/xbutil examine
System Configuration
  OS Name              : Linux
  Release              : 6.8.8
  Version              : #2 SMP PREEMPT_DYNAMIC Fri May  3 14:13:56 CDT 2024
  Machine              : x86_64
  CPU Cores            : 16
  Memory               : 94278 MB
  Distribution         : Ubuntu 22.04.3 LTS
  GLIBC                : 2.35
  Model                : F7BSC
  BIOS vendor          : American Megatrends International, LLC.
  BIOS version         : 1.04

XRT
  Version              : 2.18.0
  Branch               : HEAD
  Hash                 : c678a9469f9b20fcb9a04bbedb5c51f8473faec0
  Hash Date            : 2024-05-24 18:16:53
  XOCL                 : unknown, unknown
  XCLMGMT              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?
  AMDXDNA              : 2.18.0_20240613, 099bf0332e3d5692f98fbbf309fa9177a95f10da
  Firmware Version     : N/A

Devices present
BDF             :  Name          
---------------------------------
[0000:c5:00.1]  :  RyzenAI-npu1  

```

```
(base) mlevental@mlevental-F7BSC:/tmp/xdna-driver/build$ uname -r
6.8.8
```

makslevental, 2024-06-15:
FYI bumping to tip and building release fixes but I have no way of knowing why.

closed 2024-06-15 by makslevental

mamin506, 2024-06-17:
> FYI bumping to tip and building release fixes but I have no way of knowing why.

Hi @makslevental , I encounter "*** stack smashing detected ***: terminated" error during the development. This is because XRT and XRT plugin package are out of sync.

makslevental, 2024-06-17:
@mamin506 thanks for confirming my suspicions. Do you have any idea which buffer is overflowing?