amd/xdna-driver #133: Broken Ubuntu Version Check for Debian

xmixahlx, 2024-06-21:
While building xrt, I encounter a broken Ubuntu version check. (I use Debian Unstable.)

Build Error:

```
CMake Error at CMake/cpackLin.cmake:94 (if):
  if given arguments:

    "(" "debian" "MATCHES" "^(ubuntu)" ")" "AND" "(" "STREQUAL" "23.10" ")"

  Unknown arguments specified
Call Stack (most recent call first):
  CMake/nativeLnx.cmake:199 (include)
  CMakeLists.txt:123 (include)
```

Commenting out the code is the quick fix for me, but raising the issue for proper handling:

```
--- cpackLin.cmake.orig	2024-06-04 21:40:18.696599826 -0700
+++ cpackLin.cmake	2024-06-04 21:40:34.634035125 -0700
@@ -91,7 +91,7 @@
       uuid-dev (>= 2.27.1)")
   endif()

-  if ((${LINUX_FLAVOR} MATCHES "^(ubuntu)") AND (${LINUX_VERSION} STREQUAL "23.10"))
+#  if ((${LINUX_FLAVOR} MATCHES "^(ubuntu)") AND (${LINUX_VERSION} STREQUAL "23.10"))
     # Workaround for the following class of cpack build failure on Ubuntu 23.10
     # CMake Error at /usr/share/cmake-3.27/Modules/Internal/CPack/CPackDeb.cmake:348 (message):
     #   CPackDeb: dpkg-shlibdeps: 'dpkg-shlibdeps: error: no dependency information
@@ -104,10 +104,10 @@
     # build/Release/_CPack_Packages/Linux/DEB/xrt_202410.2.17.0_23.10-amd64/xrt directory
     # Adding an empty DEBIAN directory somehow convinces dpkg-shlibdeps to behave sanely.

-    message("-- Enable Ubuntu 23.10 cpack dpkg-shlibdeps failure workaround")
-    file(WRITE "${CMAKE_CURRENT_BINARY_DIR}/please-mantic.txt" "Workaround for cpack bug on Ubuntu 23.10")
-    install(FILES "${CMAKE_CURRENT_BINARY_DIR}/please-mantic.txt" DESTINATION "${XRT_INSTALL_DIR}/DEBIAN")
-  endif()
+#    message("-- Enable Ubuntu 23.10 cpack dpkg-shlibdeps failure workaround")
+#    file(WRITE "${CMAKE_CURRENT_BINARY_DIR}/please-mantic.txt" "Workaround for cpack bug on Ubuntu 23.10")
+#    install(FILES "${CMAKE_CURRENT_BINARY_DIR}/please-mantic.txt" DESTINATION "${XRT_INSTALL_DIR}/DEBIAN")
+#  endif()

   if (DEFINED CROSS_COMPILE)
     if (${aarch} STREQUAL "aarch64")
```

mamin506, 2024-06-22:
Thanks for reporting this and a workaround.
The broken check is cpackLin.cmake in [XRT repo](https://github.com/Xilinx/XRT). We will take a look at this and fix XRT.

xuwd1, 2024-07-25:
BTW, all XRT kernel modules (xocl.ko and xclmgmt.ko) are also broken for kernel 6.10