amd/xdna-driver #134: Build error for xrt with gcc-14 and solution

xmixahlx, 2024-06-21:
I received an error building xrt with gcc-14:

```
In file included from /home/mike/Development/MISC/xdna-driver_main/xrt/src/runtime_src/tools/xclbinutil/R>
                 from /home/mike/Development/MISC/xdna-driver_main/xrt/src/runtime_src/tools/xclbinutil/R>
/usr/include/rapidjson/document.h: In member function 'rapidjson::GenericStringRef<CharType>& rapidjson::>
/usr/include/rapidjson/document.h:319:82: error: assignment of read-only member 'rapidjson::GenericString>
  319 |     GenericStringRef& operator=(const GenericStringRef& rhs) { s = rhs.s; length = rhs.length; }

```
This issue is also described in (several) bug reports as an issue with rapidjson 1.1.0 (the current release, although very old):
https://www.mail-archive.com/devel@lists.fedoraproject.org/msg196747.html
https://bugs.gentoo.org/919374

The solution is to either patch 1.1.0 with a more recent commit:

https://github.com/Tencent/rapidjson/commit/3b2441b8.patch

... or pull rapidjson from git upstream (which is what I did to resolve after disabling the build tests):

git clone --depth=1 https://github.com/tencent/rapidjson.git rapidjson
cd rapidjson
mkdir tmpbuild
cd tmpbuild
CC=gcc-14 CXX=g++-14 cmake -DRAPIDJSON_BUILD_TESTS=OFF ..
make -j `nproc`
sudo make install

Now all of the XDNA components (linux kernel, xrt, xrt-xdna) successfully build with gcc-14.

Thanks @maxzhen and @sonals for your help.

Cheers,
Michael

keryell, 2024-06-26:
I think this is also https://github.com/Xilinx/XRT/issues/8139

keryell, 2024-06-26:
The long term solution is to use a modern library instead.

closed 2024-07-24 by maxzhen