amd/xdna-driver #163: Would add more document for use case of xdna-driver

joebnb, 2024-07-17:
I want use amd chips to model inference on some docker container，like this image

https://hub.docker.com/r/siutin/stable-diffusion-webui-docker

but im not sure how to make amd gpu work with it. 

would give some advice or document to explain how to make a cpu works docker image or linux system to extend ryzen ai support

joebnb, 2024-07-17:
https://hub.docker.com/r/intel/intel-optimized-pytorch seems intel take advantage alot

mamin506, 2024-07-17:
@gyang1099, can you please help on this?

gyang1099, 2024-07-17:
@joebnb , providing both XRT and XRT plugin packages are installed on the host system, run your docker container with additional option "--privileged -v /dev/accel:/dev/accel" to gain access to RyzenAI device, you can then run the example_noop_test or your model inference in the container

joebnb, 2024-07-18:
i using default config build get this error
```
make[4]: *** [Makefile:1934: .] Error 2
make[3]: *** [debian/rules:74: build-arch] Error 2
dpkg-buildpackage: error: make -f debian/rules binary subprocess returned exit status 2
make[2]: *** [scripts/Makefile.package:121: bindeb-pkg] Error 2
make[1]: *** [/root/linux/Makefile:1555: bindeb-pkg] Error 2
make: *** [Makefile:240: __sub-make] Error 2
```

would offering some release build by default,and write some document for use, i think many people was not know how to use this,7840 8845(this i'm using) are aleardy release over 1 year, greate chips but just few people know how to use

joebnb, 2024-07-18:
i was reading this repo commit history,maybe need more staff to help finish this,it's seems just one or two guys work on it

joebnb, 2024-07-18:
https://rocm.docs.amd.com/projects/install-on-linux/en/develop/how-to/3rd-party/pytorch-install.html#using-a-docker-image-with-pytorch-pre-installed

whats is this relation of xdna-driver  should i install rocm ? should i do mapping of  --device=/dev/kfd --device=/dev/dri  and add arguments --cap-add=SYS_PTRACE

joebnb, 2024-07-18:
[image]
and if fond there already have prebuild driver，why there was not documented or do a release using github release

joebnb, 2024-07-18:
[image]

im tired maybe should sell this and by a intel architecture machine

closed 2024-07-19 by joebnb