amd/xdna-driver #168: Help for install xdna-driver "No such device with index '0'"

joebnb, 2024-07-19:
I'm having trouble running the example after building a kernel with the newest XRT and driver for amd 8845hs.

the  operation is under PVE virtual machine

These are what i check for each step

#### Check hardware
to ensure  driver will install check the VM system requirements and host requirements

##### host check
```sh
root@pve:~# dmesg | grep -i iommu
# boot options
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.8-2-pve root=/dev/mapper/pve-root ro quiet amd_iommu=on iommu=pt nomodeset pcie_acs_override=downstream initcall_blacklist=sysfb_init vfio_iommu_type1.allow_unsafe_interrupts=1  video=efifb:off
[    0.000000] Warning: PCIe ACS overrides enabled; This may allow non-IOMMU protected peer-to-peer DMA
[    0.063268] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.8-2-pve root=/dev/mapper/pve-root ro quiet amd_iommu=on iommu=pt nomodeset pcie_acs_override=downstream initcall_blacklist=sysfb_init vfio_iommu_type1.allow_unsafe_interrupts=1 video=efifb:off
[    0.647008] iommu: Default domain type: Passthrough (set via kernel command line)
# found this
[    0.683854] pci 0000:00:00.2: AMD-Vi: IOMMU performance counters supported
[    0.683910] pci 0000:00:01.0: Adding to iommu group 0
[    0.683927] pci 0000:00:01.2: Adding to iommu group 1
[    0.683944] pci 0000:00:01.4: Adding to iommu group 2
[    0.683973] pci 0000:00:02.0: Adding to iommu group 3
[    0.683990] pci 0000:00:02.1: Adding to iommu group 4
[    0.684006] pci 0000:00:02.2: Adding to iommu group 5
[    0.684022] pci 0000:00:02.3: Adding to iommu group 6
[    0.684045] pci 0000:00:03.0: Adding to iommu group 7
[    0.684063] pci 0000:00:03.1: Adding to iommu group 8
[    0.684087] pci 0000:00:04.0: Adding to iommu group 9
[    0.684105] pci 0000:00:04.1: Adding to iommu group 10
[    0.684133] pci 0000:00:08.0: Adding to iommu group 11
[    0.684149] pci 0000:00:08.1: Adding to iommu group 12
[    0.684165] pci 0000:00:08.2: Adding to iommu group 13
[    0.684181] pci 0000:00:08.3: Adding to iommu group 14
[    0.684209] pci 0000:00:14.0: Adding to iommu group 15
[    0.684223] pci 0000:00:14.3: Adding to iommu group 15
[    0.684292] pci 0000:00:18.0: Adding to iommu group 16
[    0.684306] pci 0000:00:18.1: Adding to iommu group 16
[    0.684321] pci 0000:00:18.2: Adding to iommu group 16
[    0.684336] pci 0000:00:18.3: Adding to iommu group 16
[    0.684351] pci 0000:00:18.4: Adding to iommu group 16
[    0.684366] pci 0000:00:18.5: Adding to iommu group 16
[    0.684383] pci 0000:00:18.6: Adding to iommu group 16
[    0.684397] pci 0000:00:18.7: Adding to iommu group 16
[    0.684418] pci 0000:01:00.0: Adding to iommu group 17
[    0.684434] pci 0000:02:00.0: Adding to iommu group 18
[    0.684450] pci 0000:03:00.0: Adding to iommu group 19
[    0.684466] pci 0000:04:00.0: Adding to iommu group 20
[    0.684483] pci 0000:05:00.0: Adding to iommu group 21
[    0.684513] pci 0000:c6:00.0: Adding to iommu group 22
[    0.684530] pci 0000:c6:00.1: Adding to iommu group 23
[    0.684547] pci 0000:c6:00.2: Adding to iommu group 24
[    0.684565] pci 0000:c6:00.3: Adding to iommu group 25
[    0.684582] pci 0000:c6:00.4: Adding to iommu group 26
[    0.684600] pci 0000:c6:00.6: Adding to iommu group 27
[    0.684619] pci 0000:c7:00.0: Adding to iommu group 28
[    0.684639] pci 0000:c7:00.1: Adding to iommu group 29
[    0.684657] pci 0000:c8:00.0: Adding to iommu group 30
[    0.684677] pci 0000:c8:00.3: Adding to iommu group 31
[    0.684696] pci 0000:c8:00.4: Adding to iommu group 32
[    0.684714] pci 0000:c8:00.5: Adding to iommu group 33
[    0.684733] pci 0000:c8:00.6: Adding to iommu group 34
# found this
[    0.687914] perf/amd_iommu: Detected AMD IOMMU #0 (2 banks, 4 counters/bank).
```

#### VM check
check cpu 
```shell
root@pve:~# lscpu
[... 227 lines cut ...]
(Reading database ... 138133 files and directories currently installed.)
Preparing to unpack .../xrt_202420.2.18.0_24.04-amd64-xrt.deb ...
Unpacking xrt (2.18.0) over (2.18.0) ...
Preparing to unpack .../xrt_202420.2.18.0_24.04-amd64-xbflash.deb ...
Unpacking xrt-xbflash (2.18.0) over (2.18.0) ...
Setting up xrt (2.18.0) ...
Setting up xrt-xbflash (2.18.0) ...
Scanning processes...
Scanning linux images...

Running kernel seems to be up-to-date.

No services need to be restarted.

No containers need to be restarted.

No user sessions are running outdated binaries.

No VM guests are running outdated hypervisor (qemu) binaries on this host.
N: Download is performed unsandboxed as root as file '/home/never/xdna-driver/xrt/build/Release/xrt_202420.2.18.0_24.04-amd64-xrt.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
```

`XILINX_XRT is not set properly`  problem still

it's seems almost success, I'd appreciate any assistance.

maxzhen, 2024-07-19:
Running NPU inside a VM is not supported. I suspect that the driver can't attach properly. You can try to reload the driver and see what you get from dmesg.

joebnb, 2024-07-19:
> Running NPU inside a VM is not supported. I suspect that the driver can't attach properly. You can try to reload the driver and see what you get from dmesg.

would tell me how to reload driver

btw: after xrt-smi examin show  amdxdna version, i performed a reboot,when execute `./xrt/build/Release/opt/xilinx/xrt/bin/unwrapped/xrt-smi examin` again amdxdna was disappear

joebnb, 2024-07-19:
im using a none root user build and install driver,wired the ENV not set by driver automatic and xdna driver was gone after a reboot
```
root@tkai:~# cd /home/never/xdna-driver
root@tkai:/home/never/xdna-driver# ./xrt/build/Release/opt/xilinx/xrt/bin/xbutil examine
----------------------------------------------------------------------
                              WARNING:
                xbutil has been renamed to xrt-smi
        Please migrate to using xrt-smi instead of xbutil.

    Commands, options, arguments and their descriptions can also be
                    reported via the --help option.
----------------------------------------------------------------------
System Configuration
  OS Name              : Linux
  Release              : 6.8.0-38-generic
  Machine              : x86_64
  CPU Cores            : 4
  Memory               : 7941 MB
  Distribution         : Ubuntu 24.04 LTS
  GLIBC                : 2.39
  Model                : Standard PC (i440FX + PIIX, 1996)
  BIOS vendor          : SeaBIOS
  BIOS version         : rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org

XRT
  Version              : 2.18.0
  Branch               : HEAD
  Hash                 : 54b1a0335ef517415d17206d30365cf4a2c380d0
  Hash Date            : 2024-07-19 18:10:24
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?

Devices present
  0 devices found
root@tkai:/home/never/xdna-driver# echo $XILINX_XRT

root@tkai:/home/never/xdna-driver# export XILINX_XRT=/opt/xilinx/xrt
root@tkai:/home/never/xdna-driver# ./xrt/build/Release/opt/xilinx/xrt/bin/xbutil examine
----------------------------------------------------------------------
                              WARNING:
                xbutil has been renamed to xrt-smi
        Please migrate to using xrt-smi instead of xbutil.

    Commands, options, arguments and their descriptions can also be
                    reported via the --help option.
----------------------------------------------------------------------
System Configuration
  OS Name              : Linux
  Release              : 6.8.0-38-generic
  Machine              : x86_64
  CPU Cores            : 4
  Memory               : 7941 MB
  Distribution         : Ubuntu 24.04 LTS
  GLIBC                : 2.39
  Model                : Standard PC (i440FX + PIIX, 1996)
  BIOS vendor          : SeaBIOS
  BIOS version         : rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org

XRT
  Version              : 2.18.0
  Branch               : HEAD
  Hash                 : 54b1a0335ef517415d17206d30365cf4a2c380d0
  Hash Date            : 2024-07-19 18:10:24
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?

Devices present
  0 devices found
```

joebnb, 2024-07-19:
> Running NPU inside a VM is not supported. I suspect that the driver can't attach properly. You can try to reload the driver and see what you get from dmesg.

i rang `sudo dmesg | grep -E 'amd|error|ipu|xrt|xdna|warn|fail'` get this print
```
[    0.000000] Linux version 6.8.0-38-generic (buildd@lcy02-amd64-049) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 (Ubuntu 6.8.0-38.38-generic 6.8.8)
[    0.070367] ACPI: _OSC evaluation for CPUs failed, trying _PDC
[    0.432412] amd_pstate: the _CPC object is not present in SBIOS or ACPI disabled
[    0.709489] amdkcl: loading out-of-tree module taints kernel.
[    0.709497] amdkcl: module verification failed: signature and/or required key missing - tainting kernel
[    1.885825] [drm] amdgpu kernel modesetting enabled.
[    1.885835] [drm] amdgpu version: 6.7.0
[    1.887568] amdgpu: Virtual CRAT table created for CPU
[    1.887596] amdgpu: Topology: Add CPU node
[    1.946732] amdgpu 0000:00:10.0: amdgpu: Unable to locate a BIOS ROM
[    1.947298] amdgpu 0000:00:10.0: amdgpu: Fatal error during GPU init
[    1.947913] amdgpu 0000:00:10.0: amdgpu: amdgpu: finishing device.
[    1.980428] amdgpu: probe of 0000:00:10.0 failed with error -22
[    3.109208] kvm_amd: TSC scaling supported
[    3.109213] kvm_amd: Nested Virtualization enabled
[    3.109214] kvm_amd: Nested Paging enabled
[    3.109217] kvm_amd: Virtual VMLOAD VMSAVE supported
[    3.109218] kvm_amd: Virtual GIF supported
[    3.109218] kvm_amd: Virtual NMI enabled
[    3.109219] kvm_amd: LBR virtualization supported
[    3.122737] amdxdna 0000:00:1b.0: aie2_init: Enable PASID failed, ret -19
[    3.124764] amdxdna 0000:00:1b.0: amdxdna_probe: Hardware init failed, ret -19
```
would help analysis is have possible to make it work

xmixahlx, 2024-07-19:
Are you also using an unpatched 6.8 kernel?

maxzhen, 2024-07-19:
Or in your VM, there is no virtual IOMMU which to support PASID... We may need to disable PASID and fall back to physical memory address mode. There is something we have to do here and it is not trivial.

joebnb, 2024-07-19:
> Are you also using an unpatched 6.8 kernel?

in vm is not patched,on pve that is patched
this is pve:
```
Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.8-2-pve root=/dev/mapper/pve-root ro quiet amd_iommu=on iommu=pt nomodeset pcie_acs_override=downstream initcall_blacklist=sysfb_init vfio_iommu_type1.allow_unsafe_interrupts=1  video=efifb:off
```
this is vm
```
BOOT_IMAGE=/vmlinuz-6.8.0-38-generic root=/dev/mapper/ubuntu--vg-ubuntu--lv ro
```
but in vm immout still have output  
```
root@vm:~# sudo dmesg | grep -i iommu
# found this, iommu retured values
[    0.163549] iommu: Default domain type: Translated
[    0.163549] iommu: DMA domain TLB invalidation policy: lazy mode
```

now im trying another solution:
https://bbs.archlinux.org/viewtopic.php?id=237818

i'm not sure is it works,pve default bios is legancy bios,im trying to create a uefi bios system

joebnb, 2024-07-19:
> Or in your VM, there is no virtual IOMMU which to support PASID... We may need to disable PASID and fall back to physical memory address mode. There is something we have to do here and it is not trivial.

host machine return
```
root@pve:~# dmesg | grep -e DMAR -e IOMMU
[    0.000000] Warning: PCIe ACS overrides enabled; This may allow non-IOMMU protected peer-to-peer DMA
[    0.683854] pci 0000:00:00.2: AMD-Vi: IOMMU performance counters supported
[    0.687914] perf/amd_iommu: Detected AMD IOMMU #0 (2 banks, 4 counters/bank). 
```

vm wasn't out put any thing
```
sudo dmesg | grep -e DMAR -e IOMMU
# nothing
```

should i need rebuild kernel follow this steps or this is not necessary

```
# Assuming you have knowledge of kernel compilation,
# this is just refreshing up a few key points.

# Clone Linux source code from your favorite repository, for example
git clone git://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git

# Usually, when people compile kernel from source code, they use current config
cp /boot/config-`uname -r` <your_build_dir>/.config   # (Option step, if you know how to do it better)
# Open <your_build_dir>/.config and add "CONFIG_DRM_ACCEL=y" #Required by XDNA Driver
# Or run instead
scripts/config --file .config --enable DRM_ACCEL
scripts/config --file .config --enable AMD_IOMMU # Option step, if you know this is not set

# Use below command to build kernel packages. Once build is done, DEB packages are at the parent directory of <your_build_dir>
make -j `nproc` bindeb-pkg
# The exact names will depend on your configuration
sudo apt reinstall ../linux-headers-6.10.0_6.10.0-1_amd64.deb ../linux-image-6.10.0_6.10.0-1_amd64.deb ../linux-libc-dev_6.10.0-1_amd64.deb
```

joebnb, 2024-07-19:
> Or in your VM, there is no virtual IOMMU which to support PASID... We may need to disable PASID and fall back to physical memory address mode. There is something we have to do here and it is not trivial.

a little infomation from pve https://pve.proxmox.com/wiki/PCI_Passthrough

```
dmesg | grep -e DMAR -e IOMMU
[    0.000000] Warning: PCIe ACS overrides enabled; This may allow non-IOMMU protected peer-to-peer DMA
[    0.683854] pci 0000:00:00.2: AMD-Vi: IOMMU performance counters supported
[    0.687914] perf/amd_iommu: Detected AMD IOMMU #0 (2 banks, 4 counters/bank).

dmesg | grep 'remapping'
# is x2apic and pasid are same thing
[    0.162407] x2apic: IRQ remapping doesn't support X2APIC mode
[    0.684985] AMD-Vi: Interrupt remapping enabled

root@pve:~# dmesg | grep -i pasid
# empty
```

this is pve official pass throw manual,wish it will help for implement this

mamin506, 2024-07-20:
About manually load driver,
1. Check module info, run `modinfo amdxdna`
2. If 1) shows module details, then the driver should be installed. Then run `modprobe amdxdna` to load driver
3. Right after step 2), you can check dmesg to see if there is error. Run `dmesg`

Step 2 and 3 needs root privilege.

joebnb, 2024-07-20:
> 3\. dmesg
i rang upper command
```
modinfo amdxdna
filename:       /lib/modules/6.8.0-38-generic/updates/dkms/amdxdna.ko.zst
description:    amdxdna driver
version:        2.18.0_20240719,7e3e157ca80cef532537a4eb319109b6c05c66d3
author:         XRT Team <runtimeca39d@amd.com>
license:        GPL
srcversion:     CD1E54156086A2401FAF552
alias:          pci:v00001022d*sv*sd*bc11sc80i*
depends:        gpu-sched
retpoline:      Y
name:           amdxdna
vermagic:       6.8.0-38-generic SMP preempt mod_unload modversions
sig_id:         PKCS#7
signer:         tkai Secure Boot Module Signature key
sig_key:        5B:AE:A2:68:1A:F4:94:3D:76:E9:A9:14:70:1F:71:CC:CA:A7:2E:D7
sig_hashalgo:   sha512
signature:      02:89:61:F4:BE:B6:16:00:D9:21:FC:E6:7B:DF:06:53:AC:04:68:71:
		8E:A5:AE:59:F5:30:87:CE:38:42:EF:16:5C:22:44:37:A0:3B:93:51:
		BF:C4:63:7A:C7:0E:29:12:85:0F:E8:ED:AE:28:69:47:FB:A9:5E:97:
		1B:46:71:62:5D:B1:0D:F9:6D:DA:CF:7D:66:7C:89:5B:7D:76:2E:98:
		35:CA:D1:99:BD:5A:0A:E1:FC:43:E5:16:D5:3D:26:C2:D8:EC:35:8A:
		A1:FB:47:BA:BE:D6:61:41:56:C3:B7:4D:F9:DA:33:41:5A:26:0F:54:
		0B:27:1A:58:5E:B4:8B:D1:A4:8D:BE:47:AE:20:B5:A2:32:45:3E:AD:
		93:BC:2A:5A:91:AC:08:E2:3E:5D:EF:BA:29:A6:2F:9B:1A:6F:F1:E8:
		90:6A:54:0A:5B:31:C9:1F:F2:EC:7A:A8:27:B3:3D:49:D3:F2:5B:BE:
		32:4F:BD:2C:D0:99:49:47:C7:DB:C5:12:5E:3D:73:A3:B2:8A:6B:35:
		BE:35:DA:0E:A2:40:69:14:DC:00:3F:3D:FB:2D:C8:82:C2:D9:4B:BA:
		12:03:97:74:58:63:88:23:3F:5A:88:AD:E4:98:14:DF:66:F4:A5:07:
		8A:85:90:91:F6:50:EF:73:68:83:B2:1F:86:C7:5B:9C
parm:           iommu_mode:0 = w/ PASID (Default), 1 = wo/ PASID, 2 = Bypass (int)
parm:           priv_load:Privileged loading runtime configure (Default false) (bool)
parm:           start_col_index:Force start column, default -1 (auto select) (int)
parm:           aie2_max_col:Maximum column could be used (int)
parm:           force_cmdlist:Force use command list (Default false) (bool)
parm:           timeout_in_sec:Seconds to timeout and recovery, default 2; 0 - No TDR (int)
parm:           mailbox_polling:1:Enable mailbox polling mode (int)

modprobe amdxdna
# nothing output also trying with sudo

sudo dmesg
[    0.000000] Linux version 6.8.0-38-generic (buildd@lcy02-amd64-049) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #38-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun  7 15:25:01 UTC 2024 (Ubuntu 6.8.0-38.38-generic 6.8.8)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.8.0-38-generic root=/dev/mapper/ubuntu--vg-ubuntu--lv ro quiet amd_iommu=on iommu=pt
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000007fffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000800000-0x0000000000807fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000808000-0x000000000080afff] usable
[    0.000000] BIOS-e820: [mem 0x000000000080b000-0x000000000080bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000080c000-0x000000000080ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000810000-0x00000000008fffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000900000-0x00000000bee63fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bee64000-0x00000000bef24fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bef25000-0x00000000bf8ecfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf8ed000-0x00000000bfb6cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bfb6d000-0x00000000bfb7efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bfb7f000-0x00000000bfbfefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bfbff000-0x00000000bff1ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bff20000-0x00000000bff3ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bff40000-0x00000000bfffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[... 906 lines cut ...]
[    3.417658] audit: type=1400 audit(1721427352.402:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="busybox" pid=635 comm="apparmor_parser"
[    3.417661] audit: type=1400 audit(1721427352.402:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="Discord" pid=630 comm="apparmor_parser"
[    3.417662] audit: type=1400 audit(1721427352.402:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="cam" pid=636 comm="apparmor_parser"
[    3.417687] audit: type=1400 audit(1721427352.402:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="brave" pid=633 comm="apparmor_parser"
[    3.417724] audit: type=1400 audit(1721427352.402:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="buildah" pid=634 comm="apparmor_parser"
[    3.417781] audit: type=1400 audit(1721427352.402:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="1password" pid=629 comm="apparmor_parser"
[    3.418815] audit: type=1400 audit(1721427352.403:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="ch-checkns" pid=640 comm="apparmor_parser"
[    3.453139] amdgpu: probe of 0000:00:10.0 failed with error -22
[    3.479708] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[... cut ...]
[   74.951272] systemd-journald[363]: /var/log/journal/52892409f8ed4921a5ca2a73622451e1/user-1000.journal: Journal file uses a different sequence number ID, rotating.
```

dmesq important  summary
```
[    2.020926] amdxdna 0000:00:1b.0: aie2_init: Enable PASID failed, ret -19
[    2.021424] amdxdna 0000:00:1b.0: amdxdna_probe: Hardware init failed, ret -19

[    3.414183] amdgpu 0000:00:10.0: amdgpu: Unable to locate a BIOS ROM
[    3.414202] amdgpu 0000:00:10.0: amdgpu: Fatal error during GPU init
```
aie2_init: Enable PASID failed i have no idea to solve this,maybe @maxzhen  says it need to do something to make workaround

sashakeel, 2024-07-22:
Any update on this?
I have exactly the same issue...

joebnb, 2024-07-22:
> Any update on this? I have exactly the same issue...

no new updates. it's likely the weekend over there, so they will probably discuss and respond on Monday.

sashakeel, 2024-07-22:
Good day,

I resolved the "no such device with index 0" issue by using Linux kernel version **6.10-rc4** instead of **v6.8.0** or **v6.10**. You can download the **v6.10-rc4** mainline kernel from [here](https://kernel.ubuntu.com/mainline/v6.10-rc4/). Ensure your system is updated to this kernel version before installing the xdna-driver.

If you are manually building the Linux kernel as per the instructions in the Readme.md, use the corresponding [source code: 
v6.10-rc4](https://github.com/torvalds/linux/archive/refs/tags/v6.10-rc4.zip).

Please let me know if there is anything else I can assist you with.

joebnb, 2024-07-23:
> ./opt/xilinx/xrt/amdxdna/dkms_driver.sh --install

thanks for share your solution, i still have issues using the method you provided,it's seems this error was being resolved
```
[    2.020926] amdxdna 0000:00:1b.0: aie2_init: Enable PASID failed, ret -19
[    2.021424] amdxdna 0000:00:1b.0: amdxdna_probe: Hardware init failed, ret -19
```

but this are still exist
```
dmesg
[    3.426926] [drm] initializing kernel modesetting (IP DISCOVERY 0x1002:0x1900 0x2014:0x8001 0xC5).
[    3.426951] [drm] register mmio base: 0xC1680000
[    3.426953] [drm] register mmio size: 524288
[    3.430949] [drm] add ip block number 0 <soc21_common>
[    3.430952] [drm] add ip block number 1 <gmc_v11_0>
[    3.430955] [drm] add ip block number 2 <ih_v6_0>
[    3.430956] [drm] add ip block number 3 <psp>
[    3.430958] [drm] add ip block number 4 <smu>
[    3.430960] [drm] add ip block number 5 <dm>
[    3.430962] [drm] add ip block number 6 <gfx_v11_0>
[    3.430964] [drm] add ip block number 7 <sdma_v6_0>
[    3.430966] [drm] add ip block number 8 <vcn_v4_0>
[    3.430968] [drm] add ip block number 9 <jpeg_v4_0>
[    3.430969] [drm] add ip block number 10 <mes_v11_0>
[    3.431234] audit: type=1400 audit(1721702364.486:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="1password" pid=640 comm="apparmor_parser"
[    3.431271] audit: type=1400 audit(1721702364.486:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="QtWebEngineProcess" pid=643 comm="apparmor_parser"
[    3.431322] audit: type=1400 audit(1721702364.486:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="brave" pid=644 comm="apparmor_parser"
[    3.431368] audit: type=1400 audit(1721702364.487:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="busybox" pid=646 comm="apparmor_parser"
[    3.431400] audit: type=1400 audit(1721702364.487:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name=4D6F6E676F444220436F6D70617373 pid=642 comm="apparmor_parser"
[    3.431443] audit: type=1400 audit(1721702364.487:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="Discord" pid=641 comm="apparmor_parser"
[    3.431484] audit: type=1400 audit(1721702364.487:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="cam" pid=647 comm="apparmor_parser"
[    3.431641] audit: type=1400 audit(1721702364.487:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="buildah" pid=645 comm="apparmor_parser"
[    3.432989] audit: type=1400 audit(1721702364.488:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="ch-checkns" pid=649 comm="apparmor_parser"
[    3.438914] [drm] BIOS signature incorrect 5b 69
[    3.438936] amdgpu 0000:00:10.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0x4556
[    3.444225] [drm] BIOS signature incorrect 5b 69
[    3.444228] amdgpu 0000:00:10.0: amdgpu: Unable to locate a BIOS ROM
[    3.444248] amdgpu 0000:00:10.0: amdgpu: Fatal error during GPU init
[    3.444259] amdgpu 0000:00:10.0: amdgpu: amdgpu: finishing device.
[    3.480438] amdgpu 0000:00:10.0: probe with driver amdgpu failed with error -22
[    3.485681] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    3.485881] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    3.485971] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[    3.677497] NET: Registered PF_QIPCRTR protocol family
[    3.703423] loop0: detected capacity change from 0 to 8
```

this is VM system information
```
uname -a
Linux tkai 6.10.0-061000rc4-generic #202406161734 SMP PREEMPT_DYNAMIC Sun Jun 16 21:47:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

sudo dmesg | grep iommu
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.10.0-061000rc4-generic root=/dev/mapper/ubuntu--vg-ubuntu--lv ro quiet amd_iommu=on iommu=pt video=efifb:off
[    0.010372] Kernel command line: BOOT_IMAGE=/vmlinuz-6.10.0-061000rc4-generic root=/dev/mapper/ubuntu--vg-ubuntu--lv ro quiet amd_iommu=on iommu=pt video=efifb:off
[    0.175135] iommu: Default domain type: Passthrough (set via kernel command line)

 ./xdna-driver/xrt/build/Release/opt/xilinx/xrt/bin/xrt-smi examine
System Configuration
  OS Name              : Linux
  Release              : 6.10.0-061000rc4-generic
  Machine              : x86_64
  CPU Cores            : 8
  Memory               : 7934 MB
  Distribution         : Ubuntu 24.04 LTS
  GLIBC                : 2.39
  Model                : Standard PC (i440FX + PIIX, 1996)
  BIOS vendor          : EFI Development Kit II / OVMF
  BIOS version         : 3.20230228-4
[... 12 lines cut ...]
```

i install 6.10-rc4 kernel on vm,the host is still 6.8,because host is not ubuntu,  i'm not sure did is virtualized environment not supported.did you you also using xdna in a VM？

if you are running under a VM,could you share your pci devices?  i would like to check if there is some issue with my pci configuration.just `lspci`

in my pci there are two graphic card one is virtualized by VM other is passthrogut AMD
```
00:02.0 VGA compatible controller: Device 1234:1111 (rev 02)
00:10.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Phoenix3 (rev c5)
```

btw:i also use `make -j `nproc` bindeb-pkg` in `git://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git` this repo trying to build kernel but get this,seems some configuration is error.
```
  LD [M]  arch/x86/kvm/kvm-amd.o
  CC [M]  arch/x86/kernel/cpuid.o
  AR      arch/x86/kernel/built-in.a
  AR      arch/x86/built-in.a
make[4]: *** [Makefile:1934: .] Error 2
make[3]: *** [debian/rules:74: build-arch] Error 2
dpkg-buildpackage: error: make -f debian/rules binary subprocess returned exit status 2
make[2]: *** [scripts/Makefile.package:121: bindeb-pkg] Error 2
make[1]: *** [/home/never/linux/Makefile:1555: bindeb-pkg] Error 2
make: *** [Makefile:240: __sub-make] Error 2
```

mamin506, 2024-07-23:
@joebnb , "aie2_init: Enable PASID failed, ret -19" means iommu_dev_enable_feature() failed. This is the API of iommu module. 
The issue here is that in a VM, there is no virtual IOMMU to support PASID. Supporting PASID in IOMMU in VM is out of scope of this repo. But we will be happy to follow up once IOMMU supports PASID in a VM.

Can you please try test XDNA driver in a bare metal machine?

joebnb, 2024-07-24:
> @joebnb , "aie2_init: Enable PASID failed, ret -19" means iommu_dev_enable_feature() failed. This is the API of iommu module. The issue here is that in a VM, there is no virtual IOMMU to support [...]

when using @sashakeel suggest,the pasid error seems being fixed,but there still have some error,would you please explain what is amd gpu need locate a BIOS(is this is motherboard BIOS?),is this error caused by the system have 2 pci vga device?

```
[    3.438914] [drm] BIOS signature incorrect 5b 69
[    3.438936] amdgpu 0000:00:10.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0x4556
[    3.444225] [drm] BIOS signature incorrect 5b 69
[    3.444228] amdgpu 0000:00:10.0: amdgpu: Unable to locate a BIOS ROM
[    3.444248] amdgpu 0000:00:10.0: amdgpu: Fatal error during GPU init
[    3.444259] amdgpu 0000:00:10.0: amdgpu: amdgpu: finishing device.
[    3.480438] amdgpu 0000:00:10.0: probe with driver amdgpu failed with error -22
```

sashakeel, 2024-07-24:
@joebnb, regarding your question 

> did you also using xdna in a VM? 

No, I am using it on Bare Metal Machine.

joebnb, 2024-07-25:
ok,seems xdna driver not support VM environment,maybe i should close this issue and create new one to clearify question

closed 2024-07-25 by joebnb

joebnb, 2024-08-01:
after a reinstall OS, kernel and re compile driver on a VM, strict follow amd installation readme still 
```shell
user@vm:~$ uname -a
Linux tkai 6.10.0-061000rc4-generic #202406161734 SMP PREEMPT_DYNAMIC Sun Jun 16 21:47:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

user@vm:~$ lspci
00:00.0 Host bridge: Intel Corporation 440FX - 82441FX PMC [Natoma] (rev 02)
00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]
00:01.1 IDE interface: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II]
00:01.2 USB controller: Intel Corporation 82371SB PIIX3 USB [Natoma/Triton II] (rev 01)
00:01.3 Bridge: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 03)
00:02.0 VGA compatible controller: Device 1234:1111 (rev 02)
00:03.0 Unclassified device [00ff]: Red Hat, Inc. Virtio memory balloon
00:05.0 PCI bridge: Red Hat, Inc. QEMU PCI-PCI bridge
00:10.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Phoenix3 (rev c5)
00:11.0 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt Radeon High Definition Audio Controller
00:12.0 Ethernet controller: Red Hat, Inc. Virtio network device
00:1b.0 Signal processing controller: Advanced Micro Devices, Inc. [AMD] AMD IPU Device
00:1e.0 PCI bridge: Red Hat, Inc. QEMU PCI-PCI bridge
00:1f.0 PCI bridge: Red Hat, Inc. QEMU PCI-PCI bridge
01:01.0 SCSI storage controller: Red Hat, Inc. Virtio SCSI

user@vm:~$ sudo dmesg | grep -E 'xdna|amdgpu'
[    1.953156] amdxdna: loading out-of-tree module taints kernel.
[    1.953161] amdxdna: module verification failed: signature and/or required key missing - tainting kernel
[    2.005615] amdxdna 0000:00:1b.0: aie2_init: Enable PASID failed, ret -19
[    2.005970] amdxdna 0000:00:1b.0: amdxdna_probe: Hardware init failed, ret -19
[    3.401012] [drm] amdgpu kernel modesetting enabled.
[    3.401157] amdgpu: Virtual CRAT table created for CPU
[    3.401183] amdgpu: Topology: Add CPU node
[    3.446697] amdgpu 0000:00:10.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0x4556
[    3.451895] amdgpu 0000:00:10.0: amdgpu: Unable to locate a BIOS ROM
[    3.452195] amdgpu 0000:00:10.0: amdgpu: Fatal error during GPU init
[    3.452451] amdgpu 0000:00:10.0: amdgpu: amdgpu: finishing device.
[    3.486359] amdgpu 0000:00:10.0: probe with driver amdgpu failed with error -22

user@vm:~$ ./xdna-driver/xrt/build/Debug/opt/xilinx/xrt/bin/xrt-smi examine
System Configuration
  OS Name              : Linux
  Release              : 6.10.0-061000rc4-generic
  Machine              : x86_64
  CPU Cores            : 8
  Memory               : 7933 MB
  Distribution         : Ubuntu 24.04 LTS
  GLIBC                : 2.39
  Model                : Standard PC (i440FX + PIIX, 1996)
  BIOS vendor          : Proxmox distribution of EDK II
  BIOS version         : 4.2023.08-4

XRT
  Version              : 2.18.0
  Branch               : HEAD
  Hash                 : 54b1a0335ef517415d17206d30365cf4a2c380d0
  Hash Date            : 2024-07-26 14:28:49
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?

Devices present
  0 devices found
```