amd/xdna-driver #178: will xdna driver and xrt support running under linux VM environment?

joebnb, 2024-07-26:
there are some discuss : https://github.com/amd/xdna-driver/issues/168

i think it's a milestone of this driver, cause a lot of cloud service provider are selling GPU computing in theres VM.and if i setting up a computing system i prefer cloud native or in my vm environment.

[image]

would amd test and make xdna driver in Linux VM environment like https://libvirt.org/index.html or PVE(actually is libvirt)

maxzhen, 2024-07-27:
VM support has not been a priority since the NPU device, so far, is only available on CPU for laptops or desktops, not on servers. We'll need to look into this more and see what we can do.

joebnb, 2024-07-27:
okay,maybe a investigation needed, there are a lot of people discuss about `amd pcie passthrough` or `amd virtualization`

> https://www.google.com/search?q=amd+passthrough

btw: it was a linux driver,the use case of linux user most are not for office,videos,games but coding,server,automative are majority.pls consider

wish when i get amd ai 9 chips it was supported

joebnb, 2024-09-20:
https://github.com/isc30/ryzen-7000-series-proxmox helpful links but stll not working

maxzhen, 2025-08-02:
We have a working virtualization solution now: https://github.com/amd/virtio-npu.
Due to hardware constraint, the NPU device can't be passed through to a VM. We have developed a PV solution based on virtio.

Two known issues for now:
1. User pointer BO is not supported
2. No PASID protection inside the VM (We do have PASID protection across VMs)