amd/xdna-driver #198: make: *** No rule to make target 'package'.  Stop.

jgmelber, 2024-08-13:
This occurs on the build step `./build.sh -package` for the step [Create DEB package for existed release or debug build.](https://github.com/amd/xdna-driver?tab=readme-ov-file#steps-to-create-release-build-deb-package) top build the XDNA Driver

mamin506, 2024-08-19:
Hi @jgmelber , thanks for reporting. I cannot reproduce this issue on my system. Can you provide more dettails? What is the version of your OS. If possible, can you attach log for ./build.sh -release?  Please let me know if there is below line in your Makefile, in Release/ or Debug/.

`@echo "... package"`

NickDeBeenSAE, 2024-09-03:
I use Nobara 40 and the DNF package driver.

Would XDNA drivers work in place of the current Mesa driver I have currently installed?

I need a version of the driver that goes back to maybe 2022 as the current2024 drivers cause GPU hangs in some games.

The current biggest culprit for GPU hangs at this time is Arma Reforger.

The RDNA drivers hang on Windows and Linux.

maxzhen, 2024-09-03:
> I use Nobara 40 and the DNF package driver.
>
> Would XDNA drivers work in place of the current Mesa driver I have currently installed?
>
> I need a version of the driver that goes back to maybe 2022 as [...]

This driver is for the NPU device. Not for the GPU device. So it won't help Win or Lin.

NickDeBeenSAE, 2024-09-03:
Aaaaahhhh, ok.

Right, where is the GPU Driver Github?

maxzhen, 2024-09-04:
I believe the GPU driver on Linux has already been upstreamed. See https://github.com/torvalds/linux/tree/master/drivers/gpu/drm/amd/amdgpu

jgmelber, 2024-09-04:
I recently followed the steps to build the driver again with no issues