amd/xdna-driver #282: Module installed successfully but seeing "loading /lib/firmware/amdnpu/1502_00/npu.sbin failed with error -22"

LakshyAAAgrawal, 2024-10-09:
Hi,

I am following the guide at https://github.com/AMDResearch/Riallto/tree/main/scripts/linux to install this driver on a Minisforum UM790 Pro device. The machine has Ubuntu 24.04, with kernel 6.10.

```
$ uname -r
6.10.0-061000rc2-generic
```

I have ensured from the bios that:
1. Secure boot is disabled
2. IPU Device is enabled

The build and installation (`./setup_riallto_linux.sh <your license file>` command the above link) go through successfully. After that, when I try to run the next step "launch_jupyter.sh <notebooks directory>", I see the following error:
```
docker: Error response from daemon: error gathering device information while adding custom device "/dev/accel/accel0": no such file or directory.
```

And the output of dmesg:
```
$ sudo dmesg | grep "xdna"
[    1.632845] amdxdna: loading out-of-tree module taints kernel.
[    1.632849] amdxdna: module verification failed: signature and/or required key missing - tainting kernel
[    1.638560] amdxdna 0000:c5:00.1: Direct firmware load for amdnpu/1502_00/npu.sbin failed with error -2
[    1.638565] amdxdna 0000:c5:00.1: aie2_init: failed to request_firmware amdnpu/1502_00/npu.sbin, ret -2
[    1.638569] amdxdna 0000:c5:00.1: amdxdna_probe: Hardware init failed, ret -2
[    1.638572] amdxdna 0000:c5:00.1: probe with driver amdxdna failed with error -2
[    3.699877]  snd_rawmidi drm_buddy drm_suballoc_helper crypto_simd iwlwifi drm_ttm_helper snd_pcm cryptd snd_seq ttm snd_seq_device rapl drm_display_helper snd_pci_acp5x snd_timer cec snd_rn_pci_acp3x snd cfg80211 snd_acp_config amdxdna(OE) rc_core snd_soc_acpi i2c_algo_bit ccp snd_pci_acp3x soundcore gpu_sched k10temp i2c_piix4 amd_pmc input_leds mac_hid serio_raw sch_fq_codel msr parport_pc ppdev lp parport efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 nvme thunderbolt crc32_pclmul nvme_core r8169 xhci_pci xhci_pci_renesas nvme_auth realtek video i2c_hid_acpi wmi i2c_hid hid
[ 4060.683744] amdxdna 0000:c5:00.1: loading /lib/firmware/amdnpu/1502_00/npu.sbin failed with error -22
[ 4060.683749] amdxdna 0000:c5:00.1: Direct firmware load for amdnpu/1502_00/npu.sbin failed with error -22
[ 4060.683751] amdxdna 0000:c5:00.1: aie2_init: failed to request_firmware amdnpu/1502_00/npu.sbin, ret -22
[ 4060.683754] amdxdna 0000:c5:00.1: amdxdna_probe: Hardware init failed, ret -22
[ 4060.683756] amdxdna 0000:c5:00.1: probe with driver amdxdna failed with error -22
```

When generating the license file, I have tried using the MAC for both my wifi device and ethernet device. MAC was entered in "XXXXXXXXXXXX" format, where all characters are capital, and without ":" or "-".

Could you please help me understand what might the issue be?

Additional debug info:
```
$ sudo lspci -vd 1022:1502
c5:00.1 Signal processing controller: Advanced Micro Devices, Inc. [AMD] AMD IPU Device
        Subsystem: Device 1f4c:b002
        Flags: fast devsel, IRQ 255, IOMMU group 24
        Memory at dc900000 (32-bit, non-prefetchable) [disabled] [size=512K]
        Memory at dc9c0000 (32-bit, non-prefetchable) [disabled] [size=8K]
        Memory at 7c10900000 (64-bit, prefetchable) [disabled] [size=256K]
        Memory at dc980000 (32-bit, non-prefetchable) [disabled] [size=256K]
        Capabilities: [48] Vendor Specific Information: Len=08 <?>
        Capabilities: [50] Power Management version 3
        Capabilities: [64] Express Endpoint, MSI 00
        Capabilities: [a0] MSI: Enable- Count=1/16 Maskable- 64bit+
        Capabilities: [c0] MSI-X: Enable- Count=16 Masked-
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [2a0] Access Control Services
        Capabilities: [2d0] Process Address Space ID (PASID)
        Kernel modules: amdxdna
```

maxzhen, 2024-10-09:
It looks like the firmware is missing so that driver can't load. You might want to file the issue under Riallto repo since they know more about how things are setup.

LakshyAAAgrawal, 2024-10-09:
Thank you so much for your prompt response.

If I understand your response correctly, I wanted to mention that the firmware is available at the location:
```
$ ls -lah /lib/firmware/amdnpu/1502_00/
total 8.0K
drwxr-xr-x 2 root root 4.0K Oct  9 12:05 .
drwxr-xr-x 3 root root 4.0K Oct  8 18:45 ..
-rw-r--r-- 1 root root    0 Oct  9 11:55 npu.sbin
```

maxzhen, 2024-10-09:
Oh, OK. Then a 0 byte npu.sbin won't work. Something is wrong.

LakshyAAAgrawal, 2024-10-09:
Yes! For now I have proceeded by manually compiling the driver by cloning this repository and following the steps.

For others reference, I have created the following discussion thread https://github.com/AMDResearch/Riallto/discussions/67 on the Riallto forum.

closed 2024-10-09 by LakshyAAAgrawal