amd/xdna-driver #299: journalctl shows xdna errors on Linuxmint 22 after succesful build

MatNieuw, 2024-11-04:
linuxmint 22, AMD 8700G processor, git source from 2024-11-03.
After some minor script/cmake adjustments, XRT and xdna .DEB file build and install fine (using gdebi).
However, it doesn't work. After reboot, "sudo journalctl -b | grep xdna" shows:
===
nov 04 13:18:03 mat5-test kernel: amdxdna: loading out-of-tree module taints kernel.
nov 04 13:18:03 mat5-test kernel: amdxdna: module verification failed: signature and/or required key missing - tainting kernel
nov 04 13:18:03 mat5-test kernel: amdxdna 0000:05:00.1: enabling device (0000 -> 0002)
nov 04 13:18:03 mat5-test kernel: amdxdna 0000:05:00.1: aie2_init: Enable PASID failed, ret -19
nov 04 13:18:03 mat5-test kernel: amdxdna 0000:05:00.1: amdxdna_probe: Hardware init failed, ret -19
nov 04 13:18:05 mat5-test kernel: Modules linked in: amdgpu(+) snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi amdxcp drm_exec input_leds drm_buddy snd_seq drm_suballoc_helper drm_ttm_helper ttm snd_seq_device snd_timer drm_display_helper usblp cec snd rc_core i2c_algo_bit soundcore intel_rapl_msr intel_rapl_common edac_mce_amd kvm_amd ccp nls_iso8859_1 kvm irqbypass wmi_bmof amdxdna(OE) rapl gpu_sched i2c_piix4 k10temp mac_hid sch_fq_codel msr parport_pc ppdev lp parport efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 hid_generic usbhid uas hid usb_storage btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 dm_mirror dm_region_hash dm_log crct10dif_pclmul crc32_pclmul polyval_clmulni polyval_generic ghash_clmulni_intel nvme sha256_ssse3 sha1_ssse3 r8169 nvme_core ahci xhci_pci xhci_pci_renesas
===
and "/opt/xilinx/xrt/bin/xrt-smi examine" shows:
===
System Configuration
  OS Name              : Linux
  Release              : 6.8.0-47-generic
  Machine              : x86_64
  CPU Cores            : 16
  Memory               : 61867 MB
  Distribution         : Linux Mint 22
  GLIBC                : 2.39
  Model                : X600-ITX
  BIOS Vendor          : American Megatrends International, LLC.
  BIOS Version         : 8.01

XRT
  Version              : 2.19.0
  Branch               : HEAD
  Hash                 : 090e3faccd90abd21e59a4edbf7ed9d9c1016d0b
  Hash Date            : 2024-11-04 13:08:10
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?
  amdxdna              : 2.19.0_20241104, 330cbcf822b280ddce5c81036fc2ee6d55e9a313

Device(s) Present
  0 devices found
===
I presume that xrt-smi cannot work properly unless the xdna errors in journalctl have disappeared.
Is a later kernel version needed? Or what could be wrong?
Thanks, Mat

maxzhen, 2024-11-04:
Looks like the PASID is not enabled on your system. You may need to enable it in BIOS.

MatNieuw, 2024-11-11:
Problem solved. I was using kernel 6.8, after upgrading to kernel 6.11 (using the "mainline" utility), doing a clean, build and install from both xrt and amdxdna , the xrt-smi command now reports a device present, and no more errors in the journal. Thank you for the assistance.

MatNieuw, 2024-11-11:
Problem solved, too old a kernel used.

closed 2024-11-11 by MatNieuw