amd/xdna-driver #307: [xrt-smi] ERROR: DRM_IOCTL_AMDXDNA_SET_STATE IOCTL failed (err=13): Permission denied

Sfinx, 2024-11-13:
validate fails with 'Permission denied' from ordinary user present in video group:

```
xrt-smi validate -d 0000:6a:00.1
XRT build version: 2.19.0
Build hash: 090e3faccd90abd21e59a4edbf7ed9d9c1016d0b
Build date: 2024-11-13 18:03:48
Git branch: HEAD
PID: 211877
UID: 1000
[Wed Nov 13 16:22:04 2024 GMT]
HOST: Shiva
EXE: /opt/xilinx/xrt/bin/unwrapped/xrt-smi
[xrt-smi] ERROR: DRM_IOCTL_AMDXDNA_SET_STATE IOCTL failed (err=13): Permission denied

```

permissions looks ok:

```
ls -l /dev/dri

Permissions    Size User Group  Date Modified    Name
drwxr-xr-x        - root root   2024-11-13 12:42 .
drwxr-xr-x        - root root   2024-11-13 18:09 ..
drwxr-xr-x        - root root   2024-11-13 12:42 by-path
crw-rw----@   226,1 root video  2024-11-13 12:43 card1
crw-rw----@   226,2 root video  2024-11-13 12:43 card2
crw-rw----@ 226,128 root render 2024-11-13 12:42 renderD128
crw-rw----@ 226,129 root render 2024-11-13 12:42 renderD129

```

examine works from the same user:

```
xrt-smi examine 
System Configuration
  OS Name              : Linux
  Release              : 6.11.7-061107-generic
  Machine              : x86_64
  CPU Cores            : 16
  Memory               : 95843 MB
  Distribution         : Ubuntu 24.04.1 LTS
  GLIBC                : 2.39
  Model                : TUXEDO Sirius 16 Gen1
  BIOS Vendor          : American Megatrends International, LLC.
  BIOS Version         : APX958_V1.00A02

XRT
  Version              : 2.19.0
  Branch               : HEAD
  Hash                 : 090e3faccd90abd21e59a4edbf7ed9d9c1016d0b
  Hash Date            : 2024-11-13 18:03:48
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?
  amdxdna              : 2.19.0_20241113, 47af32f568805d6180f45883fced5277ee3a2de6
  NPU Firmware Version : 1.5.2.378

Device(s) Present
|BDF             |Name          |
|----------------|--------------|
|[0000:6a:00.1]  |RyzenAI-npu1  |
```

What can be the problem ?

maxzhen, 2024-11-13:
Yeah, it requires sudo privilege now.

Sfinx, 2024-11-14:
I see. Thanks !

closed 2024-11-14 by Sfinx

Sfinx, 2024-11-14:
BTW, it segfaults at validate:

```
[Thu Nov 14 08:21:50 2024] xrt-smi[728384]: segfault at 0 ip 00007066adfa8010 sp 000070669f7fdd20 error 4 in libxrt_coreutil.so.2.19.0[1a8010,7066ade69000+16c000] likely on CPU 2 (core 2, socket 0)
[Thu Nov 14 08:21:50 2024] xrt-smi[728383]: segfault at 30 ip 00007066adfa8013 sp 000070669fffed20 error 4
[Thu Nov 14 08:21:50 2024] Code: 48 89 fb 48 83 ec 28 64 48 8b 04 25 28 00 00 00 48 89 44 24 18 31 c0 90 e8 5d 1f fd ff 84 c0 75 19 e8 d4 1e fd ff 84 c0 75 10 <48> 8b 3b 48 8b 07 ff 50 18 eb 28 0f 1f 44 00 00 48 89 e5 48 8d 35
[Thu Nov 14 08:21:50 2024]  in libxrt_coreutil.so.2.19.0[1a8013,7066ade69000+16c000] likely on CPU 1 (core 1, socket 0)
[Thu Nov 14 08:21:50 2024] Code: 48 83 ec 28 64 48 8b 04 25 28 00 00 00 48 89 44 24 18 31 c0 90 e8 5d 1f fd ff 84 c0 75 19 e8 d4 1e fd ff 84 c0 75 10 48 8b 3b <48> 8b 07 ff 50 18 eb 28 0f 1f 44 00 00 48 89 e5 48 8d 35 19 3b 03
```

maxzhen, 2024-11-14:
We're fixing it. Please skip these test cases for now.
Just do xrt-smi validate -r quick -d for now.