amd/xdna-driver #323: Support for exposing information to `hwmon`

ThatOneCalculator, 2024-12-10:
This will allow tools like https://github.com/nokyan/resources to fully display usage statistics for the NPU, useful for when it's under load.

ref: https://github.com/nokyan/resources/issues/417

mamin506, 2025-05-19:
Hi @ThatOneCalculator , #526 start to provide per DRM client usage statistics via fdinfo.

houlz0507, 2026-04-06:
@ilot95, it seems the "package_upstream_driver" is not loaded. What is the output of "dmesg | grep amdxdna_accel_driver"?

NishadSaraf, 2026-04-06:
Hi @ilot95 and @houlz0507,

The non-fatal error `aie2_get_info: Not supported request parameter 14` has been resolved by PR #1237.

However, during further investigation on Phoenix, I found that while the sensor query (`DRM_AMDXDNA_QUERY_SENSORS`) reaches the function `aie2_get_sensors` correctly, the subsequent call to `amd_pmf_get_npu_data()` returns `-EOPNOTSUPP (-95)`. This indicates that the PMF driver on this platform does not support NPU power metrics. Phoenix NPU power reporting would require PMF driver changes to add support for this SoC.