amd/xdna-driver #324: Memory stats exposed to `fdinfo` incorrect

ThatOneCalculator, 2024-12-11:
> I don't think that supports temps, power usage, etc, but that's good to know!
> Also, unless I'm doing something wrong, it seems like memory and usage isn't reported at all, or at least incorrectly.

```
Opening /dev/accel/accel0...
Got fd: 4

Trying to read DRM stats from /proc/self/fdinfo/4:
  drm-driver:    amdxdna_accel_driver
  drm-client-id:    76
  drm-pdev:    0000:c5:00.1
  drm-engine-npu-amdxdna:    0 ns
  drm-total-memory:    0
  drm-shared-memory:    0
  drm-active-memory:    0
```

```rs
use std::{fs::File, io::{BufRead, BufReader}, os::fd::AsRawFd};

fn main() {
    println!("Opening /dev/accel/accel0...");
    match File::open("/dev/accel/accel0") {
        Ok(f) => {
            let fd = f.as_raw_fd();
            println!("Got fd: {}", fd);
            
            println!("\nTrying to read DRM stats from /proc/self/fdinfo/{}:", fd);
            match File::open(format!("/proc/self/fdinfo/{}", fd)) {
                Ok(fdinfo) => {
                    let reader = BufReader::new(fdinfo);
                    for line in reader.lines().flatten() {
                        if line.starts_with("drm-") {
                            println!("  {}", line);
                        }
                    }
                }
                Err(e) => println!("Error opening fdinfo: {}", e),
            }
        }
        Err(e) => println!("Error opening device: {}", e),
    }
}
```

_Originally posted by @ThatOneCalculator in https://github.com/nokyan/resources/issues/417#issuecomment-2533386665_

Output of `xrt-smi examine`:

```
System Configuration
  OS Name              : Linux
  Release              : 6.13.0-rc2-1-mainline-um5606-dirty
  Machine              : x86_64
  CPU Cores            : 24
  Memory               : 31196 MB
  Distribution         : Arch Linux
  GLIBC                : 2.40
  Model                : ASUS Zenbook S 16 UM5606WA_UM5606WA
  BIOS Vendor          : American Megatrends International, LLC.
  BIOS Version         : UM5606WA.312

XRT
  Version              : 2.19.0
  Branch               : master
  Hash                 : 42e5fb6f572cc94b6593cba1fca0536ed8f9d49c
  Hash Date            : 2024-12-09 22:17:48
  xocl                 : unknown, unknown
  xclmgmt              : unknown, unknown
WARNING: xclmgmt version is unknown. Is xclmgmt driver loaded? Or is MSD/MPD running?
  amdxdna              : 2.19.0_20241209, 05c3d73ef7ab93405c8866c4c0c12239d3afe5fc
  NPU Firmware Version : 1.0.0.63

Device(s) Present
|BDF             |Name          |
|----------------|--------------|
|[0000:c5:00.1]  |RyzenAI-npu4  |
```

houlz0507, 2025-01-08:
memory usage is supported. the test program opened the device and does not allocate any buffer. That is why zero usage is shown. I run 'xrt-smi validate -d -r df-bw' and 'cat /proc/.../fdinfo' in other window. I can see the valid memory usage.
# cat /proc/57340/fdinfo/3
pos:	0
flags:	0100002
mnt_id:	28
ino:	1039
drm-driver:	amdxdna_accel_driver
drm-client-id:	12
drm-pdev:	0000:c3:00.1
drm-engine-npu-amdxdna:	0 ns
drm-total-memory:	2162700 KiB
drm-shared-memory:	0
drm-active-memory:	2097156 KiB

Please modify the test program and retry. Or you may try the same xrt-smi test.

houlz0507, 2025-02-25:
Seems it is clear now? Please re-open if needed.

closed 2025-02-25 by houlz0507