amd/xdna-driver #342: Building the shim without the kernel driver

om26er, 2025-01-03:
i dont want to build the kernel driver. i only need to build the shim library so that xrt could work with AND NPU.

How do i do that?

maxzhen, 2025-01-03:
We need to add some new options to support this. Will look into it soon.

om26er, 2025-01-03:
@maxzhen do we know if the xrt shim is supposed to work with amdxdna driver in the linux kernel (linux-next right now) or is that going to need some changes in the shim code as well ?

maxzhen, 2025-01-03:
It should just work. @houlz0507 can confirm.

houlz0507, 2025-01-03:
Yes. It works fine. And the upstream driver is a little behind. E.g. import/export BO is not supported yet in upstream driver.

lamikr, 2025-01-09:
I have patched the upstream driver version to 6.13.rc5 and used that version on Fedora 40 with the xrt runtime and xrt-plugin driver available by this repo. At least the example_noop_test is able to detect the npu and test passes.

For firmwares I used the versions available in this xdna-driver repo. Btw, does anybody know how to get the files from /lib/firmware/amdnpu folder to be added automatically to kernel's initrd image on Fedora? When ever I do the "make install" for kernel, amdnpu firmwares does not get into initrd. To get amdnpu firmware files to initrd so that amdxdna driver will find them,  I need now run this dracut command always after kernel install.

`
dracut -f -i /lib/firmware /lib/firmware
`

Maybe I am missing some dracut conf file I should edit...

houlz0507, 2025-01-09:
The firmwares are upstreamed. Maybe you can try clone https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/. Then build firmware package and install it.

xanderlent, 2025-04-13:
> I have patched the upstream driver version to 6.13.rc5 and used that version on Fedora 40 with the xrt runtime and xrt-plugin driver available by this repo. At least the example_noop_test is able to [...]

Without [this upstream commit](https://github.com/torvalds/linux/commit/ecee4d0695067ae04959b121028b42a588e75370) the amdxnda module does not have the right metadata included for dracut to automatically pick up the firmware. Upgrading to a newer kernel (which automatically regenerates the initramfs with installed firmware) should fix this issue.

xanderlent, 2025-04-13:
I've proposed two possible approaches for fixing this in #472. Please take a look and let me know what you think.

maxzhen, 2025-04-13:
> Without [this upstream commit](https://github.com/torvalds/linux/commit/ecee4d0695067ae04959b121028b42a588e75370) the amdxnda module does not have the right metadata included for dracut to [...]

Or you can exclude the amdxdna driver from the ramdisk as done here: https://github.com/amd/xdna-driver/blob/8a7084ddc8a400314587945adfcda08fbc9bd083/CMake/config/postinst.in#L27
The NPU device is not required when you boot Linux anyway, so there is no need to pollute your ramdisk. Loading it a little later does not hurt.

commit 54b24620 (2025-04-14): Add SKIP_KMOD/nokmod build option (fixes #342) (#472)

This allows the project to be built without building the kernel module.
closed 2025-04-14 by maxzhen