amd/xdna-driver #363: Compile error

zamroni111, 2025-01-19:
root@ubuntu111:/var/tmp/xdna/xdna-driver/build# ./build.sh -j 1 -release
-- XRT_INSTALL_PREFIX is set /opt/xilinx
-- CMAKE_INSTALL_PREFIX=/opt/xilinx
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Looking for DRM - found at /usr 2.4.113
-- Looking for OPENCL - found at   
-- Looking for GIT - found at /usr/bin/git
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: system filesystem program_options 
-- XRT EA eula files  /var/tmp/xdna/xdna-driver/xrt/src/../LICENSE
-- Platform/Linux (ubuntu) (Kernel 6.12.6c)
-- Compiler: /usr/bin/c++ /usr/bin/cc
-- add_subdirectory(runtime_src)
-- add_subdirectory(doc)
-- add_subdirectory(xdp)
-- add_subdirectory(tools/xclbinutil)
-- Host system processor is x86_64
-- Target system processor is x86_64
-- RapidJSON found. Headers: /usr/include
-- Looking for GIT - found at /usr/bin/git
-- add_subdirectory(xocl)
-- xrt_add_subdirectory excludes xrt
-- add_subdirectory(tools/scripts)
-- add_subdirectory(core)
-- add_subdirectory(common)
-- add_subdirectory(aiebu)
-- CMAKE_SYSTEM=Linux-6.12.6c
-- Compiling AIEBU as a submodule of: /var/tmp/xdna/xdna-driver/xrt
-- CMAKE_INSTALL_PREFIX=/opt/xilinx
-- Using aie-rt headers from /var/tmp/xdna/xdna-driver/build/Release/xrt/src/runtime_src/core/common/aiebu/lib/aie-rt/driver/driver-src/include
-- Using aie-rt build from /var/tmp/xdna/xdna-driver/build/Release/xrt/src/runtime_src/core/common/aiebu
-- Using ELFIO from /var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/elf
-- OpenSSL path: /usr/lib/x86_64-linux-gnu/libssl.so;/usr/lib/x86_64-linux-gnu/libcrypto.so
-- aiebu static building enabled
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Python version: 3.10.12
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0")  
-- Boost version: 1.74.0
-- AIEBU_CPP_FILES: /var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/analyzer/reporter.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/analyzer/transaction.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/assembler/aiebu_assembler.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/assembler/assembler.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/common/aiebu_error.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/common/writer.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/elf/elfwriter.cpp;/var/tmp/xdna/xdna-driver/xrt/src/runtime_src/core/common/aiebu/src/cpp/aiebu/src/preprocessor/aie2/aie2_blob_preprocessor_input.cpp
-- Enabling build of aie-rt as submodule of aiebu
-- Build type:  Release
-- Host:    Linux/x86_64
-- Target:  Linux/x86_64
-- Machine: Generic
-- Skipping private cpack of AIEBU since compiling as a submodule
-- add_subdirectory(pcie)
-- add_subdirectory(common)
-- xrt_add_subdirectory excludes tools
-- add_subdirectory(linux)
-- add_subdirectory(emulation)
-- add_subdirectory(noop)
-- xrt_add_subdirectory excludes tools
-- add_subdirectory(python)
-- Found Python3: /usr/bin/python3 (found version "3.10.12") found components: Development Interpreter Development.Module Development.Embed 
-- Python libs version: 3.10.12
-- PYTHON_INCLUDE_PATH /usr/include/python3.10
-- Found pybind11: /usr/local/lib/python3.10/dist-packages/pybind11/include (found version "2.13.6")
-- XRT version: 2.19.0
Install components in the project: Runtime;base-dev;runtime;xrt
-- Release DEB package
-- Skipping bundling of XRT Alveo drivers with XRT package
-- Preparing OpenCL ICD xilinx.icd
-- Preparing XRT pkg-config
-- Coverity tool not found and will be skipped
-- Preparing XRT find_package
-- xrt_include excludes CMake/nativeTests.cmake
-- BUILD_REPO_ROOT: /var/tmp/xdna/xdna-driver
-- Extra driver source dir: 
-- Extra driver tar exclude rules: 
-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/xdna/xdna-driver/build/Release

real	0m0.564s
user	0m0.426s
sys	0m0.144s
[  0%] Collect all driver source code
[  0%] Build amdxdna.ko
**ERROR: modpost: "drm_gem_vmap_unlocked" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "drm_gem_shmem_vmap" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "accel_open" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "drm_gem_shmem_get_sg_table" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "pci_msix_vec_count" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "mmu_interval_notifier_insert_locked" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "drm_gem_vunmap_unlocked" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "drm_gem_shmem_create" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "drm_gem_shmem_unpin_locked" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
ERROR: modpost: "drm_gem_shmem_unpin" [/var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/amdxdna.ko] undefined!
WARNING: modpost: suppressed 25 unresolved symbol warnings because there were too many)**
make[7]: *** [scripts/Makefile.modpost:145: /var/tmp/xdna/xdna-driver/build/Release/src/driver/amdxdna/driver/amdxdna/Module.symvers] Error 1
make[6]: *** [/usr/src/linux-headers-6.12.6c/Makefile:1889: modpost] Error 2
make[5]: *** [Makefile:224: __sub-make] Error 2
make[4]: *** [Makefile:29: modules] Error 2
make[3]: *** [/var/tmp/xdna/xdna-driver/build/Release/src/driver/driver/amdxdna/Makefile:20: all] Error 2
make[2]: *** [src/driver/CMakeFiles/driver.dir/build.make:73: src/driver/amdxdna.ko] Error 2
make[1]: *** [CMakeFiles/Makefile2:2418: src/driver/CMakeFiles/driver.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

real	0m5.308s
user	0m4.763s
sys	0m0.762s
root@ubuntu111:/var/tmp/xdna/xdna-driver/build#

zamroni111, 2025-01-30:
@maxzhen  can you help on the compile eror?
how to solve the modpost error?

maxzhen, 2025-01-30:
looks like you have missing symbols from your kernel. @houlz0507 may be able to help to see what config is missing in your kernel.

houlz0507, 2025-01-31:
@zamroni111  this looks odd. The DRM_ACCEL, DRM_SCHED and DRM_GEM_SHMEM_HELPER are on in your config file, and the drm errors are still seen. Did you compile the kernel by yourself? Could you double check if attached 612e.txt is the same with /boot/config-`uname -r` on your machine?

zamroni111, 2025-01-31:
> [@zamroni111](https://github.com/zamroni111) this looks odd. The DRM_ACCEL, DRM_SCHED and DRM_GEM_SHMEM_HELPER are on in your config file, and the drm errors are still seen. Did you compile the [...]

hi @houlz0507 ,

this is the uname result.

`root@ubuntu111:/var/tmp/linux-6.12.11# uname -a
Linux ubuntu111 6.12.11e #2 SMP Thu Jan 30 21:48:58 WIB 2025 x86_64 x86_64 x86_64 GNU/Linux
root@ubuntu111:/var/tmp/linux-6.12.11# 
`

i did compile the kernel myself.
the config file attached is have exact same content as previous file.

btw, can you share the resulted Deb files for zen4 phoenix?
this can be the way forward as most users prefer to just install deb rather than having to compile.
moreover, amd usually launches new cpu gen only once a year.

[config-6.12.11e.txt](https://github.com/user-attachments/files/18626497/config-6.12.11e.txt)

zamroni111, 2025-02-01:
i found the fix, i.e. not to enable CONFIG_TRIM_UNUSED_KSYMS (Trim unused exported kernel symbols)

closed 2025-02-01 by zamroni111