amd/xdna-driver #375: How to verify installation?

sdmcclain, 2025-02-06:
Is there an easy to way to test that driver has been built and installed correctly?

sdmcclain, 2025-02-06:
Perhaps I misunderstood:

Is the [Test](https://github.com/amd/xdna-driver?tab=readme-ov-file#test) what I'm looking for? I originally assumed this was for testing changes to the source code.

keryell, 2025-02-06:
Technically the test really runs a program on the NPU. So, if it runs, that means that you have the stack working.
But the stack might have been installed in another way on your system...

steve-marmalade, 2025-02-06:
> Technically the test really runs a pre-built downloaded program on the NPU. So, if it runs, that means that you have the stack working.
> But the stack might have been installed in another way on your system...

Thanks for the detail, will close this

closed 2025-02-06 by sdmcclain