amd/xdna-driver #389: Trying amdxdna from 6.14-rc3 branch - Got noisy errors instead

waltercool, 2025-02-24:
Hi everyone,

Just tried kernel 6.14-rc3 from Linus branch last week, and got the following error.

```
[    5.512969] amdxdna 0000:c6:00.1: enabling device (0000 -> 0002)
[    5.573771] amdxdna 0000:c6:00.1: [drm] *ERROR* aie2_get_mgmt_chann_info: Invalid mbox magic 0x30ec000
[    5.573807] amdxdna 0000:c6:00.1: [drm] *ERROR* aie2_hw_start: firmware is not alive
[    5.652923] amdxdna 0000:c6:00.1: [drm] *ERROR* aie2_smu_exec: smu cmd 4 failed, 0xff
[    5.652955] amdxdna 0000:c6:00.1: [drm] *ERROR* aie2_smu_fini: Power off failed, ret -22
[    5.652976] amdxdna 0000:c6:00.1: [drm] *ERROR* aie2_init: start npu failed, ret -22
[    5.653244] amdxdna 0000:c6:00.1: [drm] *ERROR* amdxdna_probe: Hardware init failed, ret -22
[    5.653266] amdxdna 0000:c6:00.1: probe with driver amdxdna failed with error -22
```

```
$ amdgpu_top --xdna
There are no the XDNA NPU devices found.
```

System: Linux kernel 6.14-rc3 from Linus branch, booted with dracut and systemd.
Hardware: AMD Ryzen 7 7840HS w/ Radeon 780M Graphics
Manufacturer: Framework 16.

houlz0507, 2025-02-25:
@waltercool , It seems incorrect firmware is used. How did you install the firmware? could you run
 "tree -s /lib/firmware/amdnpu/"
and post the size of firmware installed?

waltercool, 2025-02-26:
Aha, I can confirm the issue now.

npu.sbin doesn't exist at latest Linux Firmware tag, it exists "npu.sbin.1.5.2.380" https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/amdnpu/1502_00

Likely had a soft link using an old version at early phase of the driver

Many thanks! It works by doing a hard link to that npu.sbin

closed 2025-02-26 by waltercool