amd/xdna-driver #421: Unable to load kernel driver with 6.14-rc7

iamkroot, 2025-03-18:
I have compiled the latest rc from https://aur.archlinux.org/packages/linux-mainline (and enabled the `CONFIG_DRM_ACCEL_AMDXDNA` config)

However, on startup, the driver load fails.

This is the log output:

```log
~ ❯ journalctl -b | grep amdxdna
Mar 18 13:11:52 altair kernel: amdxdna 0000:64:00.1: Direct firmware load for amdnpu/17f0_10/npu.sbin failed with error -2
Mar 18 13:11:52 altair kernel: amdxdna 0000:64:00.1: [drm] *ERROR* aie2_init: failed to request_firmware amdnpu/17f0_10/npu.sbin, ret -2
Mar 18 13:11:52 altair kernel: amdxdna 0000:64:00.1: [drm] *ERROR* amdxdna_probe: Hardware init failed, ret -2
Mar 18 13:11:52 altair kernel: amdxdna 0000:64:00.1: probe with driver amdxdna failed with error -2
~ ❯ l /lib/firmware/amdnpu/17f0_10
.rw-r--r-- 377k root 18 Mar 13:10 npu.sbin
.rw-r--r-- 377k root 18 Mar 13:10 npu.sbin.1.0.0.63
.rw-r--r--  73k root 18 Mar 08:45 npu.sbin.1.0.0.63.zst
.rw-r--r--  73k root 18 Mar 08:45 npu.sbin.1.0.0.63.zst.zst
lrwxrwxrwx    - root 18 Mar 08:45 npu.sbin.zst -> npu.sbin.1.0.0.63.zst
~ ❯ diff /lib/firmware/amdnpu/17f0_10/npu.sbin /lib/firmware/amdnpu/17f0_10/npu.sbin.1.0.0.63 
~ ❯ uname -r                                        
6.14.0-rc7-1-mainline
```

iamkroot, 2025-03-18:
More info about the files:
1. I have also built `linux-firmware-git` package this morning
2. The `npu.sbin.1.0.0.63.zst` file was already present. The zstd configs (`CONFIG_FW_LOADER_COMPRESS_ZSTD` and `CONFIG_ZSTD_DECOMPRESS`) are on by default.
3. Just in case, I manually unpacked the archive to `npu.sbin.1.0.0.63` and initially created a symlink `npu.sbin` pointing to this file. Same error.
4. Taking note from #389 I downloaded the `npu.sbin` file directly from the kernel.org, but turns out it is the same as the one inside the `zst`. So no difference there either.

Not sure why `request_firmware` isn't able to find the file

maxzhen, 2025-03-18:
This is on Ubuntu or Fedora?

iamkroot, 2025-03-18:
Ah, I should have clarified. This is on Arch Linux. I assumed now that the driver is in mainline, it would load fine with any distro.

tomchiverton, 2025-04-08:
I think I have this same issue with a Framework 13" AMD under the Ubuntu 25.04 beta. How would I check " if both firmware and driver are inside initram "?

```
falken@frame:~$ ls -laht /usr/lib/firmware/amdnpu/1502_00/
total 80K
drwxr-xr-x 2 root root 4.0K Apr  6 10:40 .
drwxr-xr-x 5 root root 4.0K Apr  6 10:40 ..
-rw-r--r-- 1 root root  72K Mar 17 12:59 npu.sbin.1.5.2.380.zst
lrwxrwxrwx 1 root root   22 Mar 17 12:59 npu.sbin.zst -> npu.sbin.1.5.2.380.zst
falken@frame:~$ journalctl |grep amdn|tail
Apr 05 12:55:45 frame kernel: amdxdna 0000:c2:00.1: Direct firmware load for amdnpu/1502_00/npu.sbin failed with error -2
falken@frame:~$ dpkg -l|grep linux-firm
ii  linux-firmware                                      20250317.git1d4c88ee-0ubuntu1              amd64        Firmware for Linux kernel drivers

```

mamin506, 2025-04-10:
Below error is printing by request_firmware() API from firmware loader.
`Mar 18 13:11:52 altair kernel: amdxdna 0000:64:00.1: Direct firmware load for amdnpu/17f0_10/npu.sbin failed with error -2`

The firmware loader has a list of fw_path to find specific firmware.
If your kernel has enabled dynamic debug (CONFIG_DYNAMIC_DEBUG = y). Please do below step to further narrow down.

``` shell
# Assume all below are running with root privilege

$ echo "func fw_get_filesystem_firmware +pf" > /sys/kernel/debug/dynamic_debug/control
$ modprobe amdxdna

# Check dmesg log. If your setup is correct, look for below debug messages. You are expecting to see multiple messages.
# "loading %s failed for no such file or directory." The %s will be the full path + filename that firmware loader is looking for.

$ echo "func fw_get_filesystem_firmware -pf" > /sys/kernel/debug/dynamic_debug/control
```

tomchiverton, 2025-04-13:
Since Ubuntu 25.04 beta upgraded to linux-image-6.14.0-15-generic and linux-firmware 20250317.git1d4c88ee-0ubuntu1 I see this at boot time, so looks like resolved

```
$ journalctl |grep xdna|tail -n 2
Apr 12 18:53:56 frame kernel: amdxdna 0000:c2:00.1: enabling device (0000 -> 0002)
Apr 12 18:53:56 frame kernel: [drm] Initialized amdxdna_accel_driver 0.0.0 for 0000:c2:00.1 on minor 0
```

closed 2025-04-13 by maxzhen