amd/xdna-driver #423: ModuleNotFoundError: No module named 'pyxrt'

munawarkholil, 2025-03-18:
File "/home/<user>/xdna-driver/xrt/src/../tests/python/200_binding/200_main.py", line 18, in <module>
    import pyxrt as pp
ModuleNotFoundError: No module named 'pyxrt'

when executing $ ./build.sh -npu -opt

maxzhen, 2025-03-18:
Please share the full build log and the version of Linux for us to reproduce the issue.

munawarkholil, 2025-03-18:
https://gist.github.com/munawarkholil/11bb0e0e4b409ac527168b1d1317221b

maxzhen, 2025-03-18:
what version of linux? do you happen to have a version of xrt installed in your build system?

munawarkholil, 2025-03-18:
[image]
Linux aigen 6.11.0-19-generic #19~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 17 11:51:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

I am using ubuntu lts 24.04 and following the guide in readme.md, am I missing something?

munawarkholil, 2025-03-18:
[image]

maxzhen, 2025-03-18:
I just tried building on my Ubuntu 24.04 and don't see any issue. Since we're using XRT as a git submodule, maybe you can repost the issue under https://github.com/Xilinx/XRT?

mamin506, 2025-04-09:
I encounter same issue. Clean build xrt submodule save me.
``` shell
$ ./build.sh -clean
$ ./build.sh -npu -opt
```

closed 2025-04-12 by munawarkholil