amd/xdna-driver #513: Framework 13 AMD Ryzen 9 AI; amdxdna_drm_submit_execbuf: Invalid arg bo count 0

brycereitano, 2025-05-06:
I'm running into errors when attempting to execute tests or validate the installation of the XRT shim. Replicated on both an an Ubuntu Live USB and on Arch following similar instructions in #480.

#### System Info
- **OS**: Ubuntu 25.04 (Linux 6.14.0-15-generic)
- **CPU**: AMD Ryzen 9 AI 370 (Framework 13)
- **XDNA Version**: `26f95ad62c293fa186ce4b6da047018d628c2e23`
- **XRT Version**: `d5835aaa7fcdbcb749f4d837d9a0a605c1a4d312` (submodule)

#### xrt-smi examine
```
System Configuration
  OS Name              : Linux
  Release              : 6.14.0-15-generic
  Machine              : x86_64
  CPU Cores            : 24
  Memory               : 95835 MB
  Distribution         : Ubuntu 25.04
  GLIBC                : 2.41
  Model                : Laptop 13 (AMD Ryzen AI 300 Series)
  BIOS Vendor          : INSYDE Corp.
  BIOS Version         : 03.03

XRT
  Version              : 2.19.0
  Branch               : HEAD
  Hash                 : d5835aaa7fcdbcb749f4d837d9a0a605c1a4d312
  Hash Date            : 2025-05-06 04:09:46
  amdxdna              : unknown, unknown
  virtio-pci           : unknown, unknown
  NPU Firmware Version : 1.0.0.63

Device(s) Present
|BDF             |Name          |
|----------------|--------------|
|[0000:c2:00.1]  |RyzenAI-npu4  |
```

## Steps

1. Boot onto Ubuntu Live USB
2. Clone the xDNA driver and XRT from GitHub:
   ```bash
   git clone https://github.com/amd/xdna-driver.git
   cd xdna-driver
   git submodule update --init --recursive --progress
   ```
3. Build and install XRT with NPU support:
   ```bash
   sudo ./tools/amdxdna_deps.sh

   cd xrt/build/
   ./build.sh -npu -opt -j 8
   sudo apt reinstall ./Release/xrt_202510.2.19.0_25.04-amd64-base.deb
   sudo apt reinstall ./Release/xrt_202510.2.19.0_25.04-amd64-base-dev.deb # required for example build
   sudo apt reinstall ./Release/xrt_202510.2.19.0_25.04-amd64-npu.deb # tried with and without
   ```
4. Build and install XDNA XRT Shim:
   ```bash
   cd ../../build
   ./build.sh -release -nokmod -j 8
   ./build.sh -package
   sudo apt reinstall ./Release/xrt_plugin.2.19.0_ubuntu25.04-x86_64-amdxdna.deb 
   ```
5. Build and run example:
   ```bash
   source /opt/xilinx/xrt/setup.sh
   ./build.sh -example
   ./example_build/example_noop_test ../tools/bins/1502_00/validate.xclbin
   # See output below
   ```

#### Example Output
```
$ ./example_build/example_noop_test ../tools/bins/1502_00/validate.xclbin
Host test code start...
Host test code is creating device object...
Host test code is loading xclbin object...
Host test code is creating kernel object...
Host test code kernel name: DPU_PDI_0
Host code is registering xclbin to the device...
Host code is creating hw_context...
Host test code is creating kernel object...
Host test code allocate buffer objects...
Host test code sync buffer objects to device...
Host test code iterations (~10 seconds): 70000
ERROR: Caught exception: DRM_IOCTL_AMDXDNA_EXEC_CMD IOCTL failed (err=-22): Invalid argument
TEST FAILED!
```

This produced the following dmesg output:
```
[ 1773.017860] amdxdna 0000:c2:00.1: [drm] *ERROR* amdxdna_drm_submit_execbuf: Invalid arg bo count 0
```

## Aux Information

#### xrt-smi validate
```bash
# /opt/xilinx/xrt/bin/xrt-smi validate
Validate Device           : [0000:c2:00.1]
    Platform              : RyzenAI-npu4
    Power Mode            : Default
-------------------------------------------------------------------------------
[                    ]: Running Test... < 0s >
terminate called after throwing an instance of 'xrt_core::system_error'
  what():  DRM_IOCTL_AMDXDNA_CONFIG_CTX IOCTL failed (err=-95): Operation not supported
/opt/xilinx/xrt/bin/unwrapped/loader: line 61: 25690 Aborted 
```

This did not produce any more information in dmesg.

#### dmesg loading
```
[   11.653928] amdxdna 0000:c2:00.1: enabling device (0000 -> 0002)
[   11.743152] [drm] Initialized amdxdna_accel_driver 0.0.0 for 0000:c2:00.1 on minor 0
```

houlz0507, 2025-05-06:
Need to update in-tree driver.

houlz0507, 2025-05-07:
Revert the shim change to enable current in-tree driver for now. PR: https://github.com/amd/xdna-driver/pull/516

brycereitano, 2025-05-07:
Fantastic, both the examples build and pass on the new branch. 🎉 Thank you for the quick turnaround!

### example run
```
# ./example_build/example_noop_test /opt/xilinx/xrt/amdxdna/bins/1502_00/validate.xclbin
Host test code start...
Host test code is creating device object...
Host test code is loading xclbin object...
Host test code is creating kernel object...
Host test code kernel name: DPU_PDI_0
Host code is registering xclbin to the device...
Host code is creating hw_context...
Host test code is creating kernel object...
Host test code allocate buffer objects...
Host test code sync buffer objects to device...
Host test code iterations (~10 seconds): 70000
Host test microseconds: 4610845
Host test average latency: 65 us/iter
TEST PASSED!
```

### xrt-smi examine
Running successfully on Arch Linux as well!
```
# /opt/xilinx/xrt/bin/xrt-smi examine
System Configuration
  OS Name              : Linux
  Release              : 6.14.5-arch1-1
  Machine              : x86_64
  CPU Cores            : 24
  Memory               : 95831 MB
  Distribution         : Arch Linux
  GLIBC                : 2.41
  Model                : Laptop 13 (AMD Ryzen AI 300 Series)
  BIOS Vendor          : INSYDE Corp.
  BIOS Version         : 03.03

XRT
  Version              : 2.19.0
  Branch               : HEAD
  Hash                 : d5835aaa7fcdbcb749f4d837d9a0a605c1a4d312
  Hash Date            : 2025-05-06 20:33:23
  amdxdna              : unknown, unknown
  virtio-pci           : unknown, unknown
  NPU Firmware Version : 1.0.0.63

Device(s) Present
|BDF             |Name          |
|----------------|--------------|
|[0000:c2:00.1]  |RyzenAI-npu4  |
```

mamin506, 2025-05-15:
Issue resolved. Close.

closed 2025-05-15 by mamin506