amd/xdna-driver #629: create_ctx becomes slower when auto suspend/resume is enabled

Yu-Zhewen, 2025-07-03:
PR #547 enabled auto suspend/resume, which causes a performance regression (+70ms) for PCI devices during hardware context creation (create_ctx).

Setting `autosuspend_ms < 0` does not resolve the issue. 

It might be because the following PM functions are still invoked regardless of the `autosuspend_ms` value?

https://github.com/amd/xdna-driver/blob/87b3920860a21f8ed52b69045c24c35a80b19792/src/driver/amdxdna/amdxdna_pm.c#L60-L75

If these PM function calls are indeed contributing significant overhead, would it be possible to add an additional flag or logic to allow users to explicitly control whether they are executed, rather than relying solely on whether the device is PCI or not?

maxzhen, 2025-07-04:
Why would context creation latency ever matter? It should never be in critical path.

Yu-Zhewen, 2025-07-07:
I agree it might not matter much for typical production use cases, but from a performance regression and benchmarking standpoint, it’s a bit concerning to see ~70ms added by the suspend/resume logic.

In tools like [iree-amd-aie](https://github.com/nod-ai/iree-amd-aie), which include many performance benchmarks, a new context is often created for each test.  This overhead accumulates in CI, increasing total runtime and affecting benchmark reliability, especially for some tests which were expected to run in <0.1ms.

maxzhen, 2025-08-02:
Context creation should never ever be included in critical path or included in any benchmark measurement. By design, you create context once during setup and destroy it after you're done with NPU device.

closed 2025-08-02 by maxzhen