amd/xdna-driver #713: Support for power reading

loostrum, 2025-08-27:
I'm very much interested in measuring the energy efficiency of the Strix Point NPU I have available here, however I noticed that the driver does not support reading power data yet, it simply returns `UINT32_MAX` as the power value. Is there a timeline for adding support for this to the xdna driver?

csbnw, 2026-03-12:
Would this also work for Strix Point NPU's?

csbnw, 2026-03-16:
For later reference, on the Strix Halo, this is what the power profile looks like when running `xrt-smi validate -r gemm`:
[image]

For the throughput test, absolute power is lower, but there is also a ramp-up visible:
[image]